OpenAgricultureFoundation/openag_brain #1: couchdb views are not copied during install

serein7, 2016-05-04:
I've noticed that after:
`pip3 install .`
no **js** files from _openag/brain/core/_design/_  subdirectories are copied and as a result db_init is failing to create corresponding couchdb views.
to fix this I had to create MANIFEST.in file in the project root directory (where setup.py is located)
`global-include *.js`
and modify setup.py with an extra parameter during setup function call:
`include_package_data=True`

LeonChambers, 2016-05-04:
Cool, I suspected that this might happen but hadn't tested it yet. Could you submit a pull request with the fix?

serein7, 2016-05-04:
Sure. Here it is https://github.com/OpenAgInitiative/openag-brain/pull/2

closed 2016-05-04 by LeonChambers