OpenAgricultureFoundation/openag_brain #3: installation instruction should mention dependency on openag-pymata-aio

serein7, 2016-05-04:
In my case execution of `python3 -m openag.brain.core.init_db` right after `git clone ...` was failing due missing openag_pymata_aio.pymata3 in modules/arduino.py
First I had to manually install openag_pymata_aio.pymata3

> cd ..
> git clone https://github.com/OpenAgInitiative/openag-pymata-aio.git
> cd openag-pymata-aio
> pip3 install .
> cd ../openag-brain

and this fixed the issue.

I'm new to python stack, so I can only guess that this might somehow be resolved in more elegant way through setup.py `install_requires` parameter or similar.

LeonChambers, 2016-05-04:
Yea, the best way to fix it would be to add `openag-pymata-aio` to the `install_requires` parameter, but we would have to register `openag-pymata-aio` in PyPI first for that to work, which we should probably do soon anyway.

serein7, 2016-05-05:
@LeonChambers, does current code suppose to register some Modules as well? After init_db I see that there are some records in module_type, but module one is an empty.
And when I start run_module I see:

> Running 0 modules
> Listening for requests on http://localhost:5000/

Am I correct assuming that it is stil work in progress and no wire code exists to proxy module API through openag-brain?

LeonChambers, 2016-05-09:
Right now you have to register the modules manually through the web interface for CouchDB. The `models.py` file in `openag.brain.core` defines what fields are required for Modules. You can also define ModuleConnections which connect an output of one module to an input of another.

serein7, 2016-05-10:
@LeonChambers , thank you! I see that in latest commit you've included test fixture as well, so I got all this running.

Now I guess I need to figure out that Firmata thing on Arduino to test it working between devices.

LeonChambers, 2016-05-10:
Yep. I don't really know the firmware setup works at the moment though. @jakerye might be able to help with that

serein7, 2016-05-11:
Yes, it looks like at least openag-firmata is not compilable. I created an issue in its repository.

serein7, 2016-05-14:
Just an update. I was able to setup a **brain** with arduino running **firmata** code and **dht22** sensor connected. Some easy fix on openag-firmata is needed and few manual records added in couchdb for arduino, dht22 and connections. Here are some details https://github.com/OpenAgInitiative/openag-firmata/issues/1

closed 2016-05-25 by serein7