OpenAgricultureFoundation/openag_brain #5: Running on laptop results in error

gordonbrander, 2016-05-10:
Following the instructions in the README, I get this error when running `python3 -m openag.brain.core.init_db`.

```bash
Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/openag/brain/core/init_db.py", line 35, in <module>
    shutil.copy(config_file_path, "/etc/couchdb/default.d/openag.ini")
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/shutil.py", line 235, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/shutil.py", line 115, in copyfile
    with open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: '/etc/couchdb/default.d/openag.ini'
```

LeonChambers, 2016-05-10:
Do you have couchdb installed? What OS are you running? On Linux, couchdb stores configuration files in `/etc/couchdb/`, but it might be different for other operating systems.

gordonbrander, 2016-05-10:
@LeonChambers 

- I have CouchDB installed
- OSX 10.11.4

I tried copying the openag.ini file from `openag/brain/core/openag.ini`, but no luck:

```
Traceback (most recent call last):
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/openag/brain/core/init_db.py", line 40, in <module>
    server.create(db_name.value)
  File "/usr/local/lib/python3.5/site-packages/couchdb/client.py", line 206, in create
    self.resource.put_json(name)
  File "/usr/local/lib/python3.5/site-packages/couchdb/http.py", line 549, in put_json
    **params)
  File "/usr/local/lib/python3.5/site-packages/couchdb/http.py", line 564, in _request_json
    headers=headers, **params)
  File "/usr/local/lib/python3.5/site-packages/couchdb/http.py", line 560, in _request
    credentials=self.credentials)
  File "/usr/local/lib/python3.5/site-packages/couchdb/http.py", line 276, in request
    conn = self.connection_pool.get(url)
  File "/usr/local/lib/python3.5/site-packages/couchdb/http.py", line 486, in get
    conn.connect()
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/http/client.py", line 826, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/socket.py", line 711, in create_connection
    raise err
  File "/usr/local/Cellar/python3/3.5.1/Frameworks/Python.framework/Versions/3.5/lib/python3.5/socket.py", line 702, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 61] Connection refused
```

LeonChambers, 2016-05-10:
Could you post the output of `couchdb -c` so I can see where couchdb puts config files on OS X?

This new error suggests that either couchdb isn't running or httpd is configured incorrectly. Does the web interface work (http://localhost:5984/_utils)?

gordonbrander, 2016-05-10:
```bash
$ couchdb -c
/usr/local/etc/couchdb/default.ini
/usr/local/etc/couchdb/local.ini
```

LeonChambers, 2016-05-10:
Does the folder `/usr/local/etc/couchdb/default.d` exist?

gordonbrander, 2016-05-11:
@LeonChambers yes

LeonChambers, 2016-05-11:
Ok, so the `FileNotFoundError` should be fixed now. I'll see if I can replicate the `ConnectionRefusedError` on a fresh install.

LeonChambers, 2016-05-16:
I was unable to replicate the `ConnectionRefusedError` on a fresh install of Ubuntu Server, so this is probably something platform-specific. Could you post the output of `curl localhost:5984/_config/httpd`?

closed 2016-05-19 by LeonChambers