OpenAgricultureFoundation/openag_brain #6: Error

edcafenet, 2016-05-11:
Hi! I get this error when running python3 -m openag.brain.core.init_db

```
Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/openag/brain/core/init_db.py", line 40, in <module>
    server.create(db_name.value)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/client.py", line 206, in create
    self.resource.put_json(name)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/http.py", line 549, in put_json
    **params)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/http.py", line 564, in _request_json
    headers=headers, **params)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/http.py", line 560, in _request
    credentials=self.credentials)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/http.py", line 276, in request
    conn = self.connection_pool.get(url)
  File "/usr/local/lib/python3.4/dist-packages/couchdb/http.py", line 486, in get
    conn.connect()
  File "/usr/lib/python3.4/http/client.py", line 834, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python3.4/socket.py", line 512, in create_connection
    raise err
  File "/usr/lib/python3.4/socket.py", line 503, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```
This is what happens when I run _couchdb -c_
```

/etc/couchdb/default.ini
/etc/couchdb/default.d/openag.ini
/etc/couchdb/local.ini
```
My _/etc/couchdb/default.d/openag.ini_ looks like this:
```
[httpd_global_handlers]
_openag = {couch_httpd_proxy, handle_proxy_req, <<"http://10.211.55.5:5984">>}

```

I have my couchdb instance running in _http://10.211.55.5:5984_

LeonChambers, 2016-05-16:
I'm guessing this is because of how httpd is configured. Could you post the output of `curl localhost:5984/_config/httpd`?

serein7, 2016-05-18:
@edcafenet do you run openag-brain from same box as couchdb, i.e. at 10.211.55.5?

edcafenet, 2016-05-18:
Yes! CouchDB and openag-brain are running on the same machine (10.211.55.5)

serein7, 2016-05-18:
@edcafenet than can you try to change the ip to 127.0.0.1 (i.e. localhost). I believe couchdb is binding itself only to localhost by default

serein7, 2016-05-18:
or, if for some reason you need to bind to this address, please, change /etc/couchdb/local.ini config file to set bind_address to your ip or 0.0.0.0. Here are more details http://stackoverflow.com/questions/5765028/getting-started-with-couchdb

serein7, 2016-05-18:
It may help to understand better if you try to open in browser:
1. http://localhost:5984/
2. http://10.211.55.5:5984/
One of those would probably give you ERR_CONNECTION_REFUSED. Which one?

LeonChambers, 2016-05-18:
@edcafenet the file `/etc/couchdb/default.d/openag.ini` tells couchdb to proxy the server run by the `run_modules` script, so the last line should say `_openag = {couch_httpd_proxy, handle_proxy_req, <<"http://localhost:5000">>}`, which is the default. As @serein7 mentioned, `/etc/couchdb/local.ini` is probably what you should be looking at the debug this because the `[httpd]` section defines the port that couchdb binds to to listen for requests.

edcafenet, 2016-05-19:
As @serein7 suggested I changed my /etc/couchdb/local.ini config file to set bind_address to 0.0.0.0. 
This fixed the previous error. However, I am getting this stuff python error instead:

`Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/openag/brain/core/init_db.py", line 59, in <module>
    modules_path = os.path.dirname(modules.__file__)
AttributeError: 'module' object has no attribute '__file__'`

serein7, 2016-05-19:
@edcafenet I see that in recent commits this code has changed. I would do`git pull`and try again. The chances are you still will have an error, but different one :)

edcafenet, 2016-05-19:
@serein7 You were right. Here you have:

`Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/openag/brain/core/init_db.py", line 76, in <module>
    py_mod = import_module(py_mod_name)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "./openag/brain/modules/arduino.py", line 2, in <module>
    from openag_pymata_aio.pymata3 import PyMata3
ImportError: No module named 'openag_pymata_aio'`

LeonChambers, 2016-05-19:
@edcafenet See #3 for the fix

edcafenet, 2016-05-19:
OK guys. I believe we are one step closer:

`Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.4/dist-packages/openag/brain/core/init_db.py", line 76, in <module>
    py_mod = import_module(py_mod_name)
  File "/usr/lib/python3.4/importlib/__init__.py", line 109, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 2254, in _gcd_import
  File "<frozen importlib._bootstrap>", line 2237, in _find_and_load
  File "<frozen importlib._bootstrap>", line 2226, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1200, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1129, in _exec
  File "<frozen importlib._bootstrap>", line 1471, in exec_module
  File "<frozen importlib._bootstrap>", line 321, in _call_with_frames_removed
  File "./openag/brain/modules/arduino.py", line 2, in <module>
    from openag_pymata_aio.pymata3 import PyMata3
  File "/usr/local/lib/python3.4/dist-packages/openag_pymata_aio/pymata3.py", line 21, in <module>
    from .pymata_core import PymataCore
  File "/usr/local/lib/python3.4/dist-packages/openag_pymata_aio/pymata_core.py", line 348
    async def start_aio(self):
            ^
SyntaxError: invalid syntax`

LeonChambers, 2016-05-19:
This is because you are using python 3.4. It should work in python 3.5

edcafenet, 2016-05-19:
Thank you guys. I think I got it working. 

Now, I will focus on trying to send and retrieve information from couchdb.

closed 2016-05-19 by LeonChambers