OpenAgricultureFoundation/openag_brain #7: Query endpoint to get ID of currently running recipe

gordonbrander, 2016-05-17:
We'll need an API endpoint that allows us to query ID of currently running recipe on Raspberry Pi. I don't see a design document for this endpoint in https://github.com/OpenAgInitiative/openag-brain/tree/master/openag/brain/core/_design. Am I missing something?

LeonChambers, 2016-05-17:
The `latest` view in `environmental_data_point` could be used to do this. When a recipe is started, it creates a data point of type RECIPE_START saying which recipe was started. The RecipeHandler module does this [here](https://github.com/OpenAgInitiative/openag-brain/blob/master/openag/brain/modules/recipe_handler.py#L74) to check if a recipe is already running when it is initialized.

gordonbrander, 2016-05-17:
To construct the query, I need to know the environment ID:

`/environmental_data_point/_design/openag/_view/latest?key=[<environment id>, "EnvironmentalVariable.RECIPE_START", "measured"]`

Do we plan to have a design document for this in the `environment` database?

LeonChambers, 2016-05-17:
No, you would just query the `environment` database to see what environments there are and send the ID of the one in which you want the start the recipe. In the food computers, there will probably only ever be 1 environment though.

closed 2016-05-19 by LeonChambers