OpenAgricultureFoundation/openag_brain #22: Error when trying to install openag_brain

Mavni, 2016-06-10:
Hi guys!

I'm trying to follow your instructions on how to install openag_brain on my R pi running Debian, but i cannot seem to get past "Pip3 install ."
I receive this error:

root@raspberrypi:/home/pi/openAG/openag_brain#  pip3 install .
Processing /home/pi/openAG/openag_brain
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-90yrosaf-build/setup.py", line 7, in <module>
        raise RuntimeError("openag_brain requires Python 3.5 or newer")
    RuntimeError: openag_brain requires Python 3.5 or newer

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-90yrosaf-build/

I installed Python3, python -V gives me "Python 3.5.1"

What am I missing?

I'm pretty new with Python, so sorry if this is a noobish question, but I searched all over the place and tried all kinds of PATH related solutions to no avail

Thanks in advance!
Michael

LeonChambers, 2016-06-11:
Hm, this seems weird. What happens if you run `pip3 --version`?

Mavni, 2016-06-11:
root@raspberrypi:/home/pi/openAG/openag_brain# pip3 -V
pip 8.1.2 from /usr/local/lib/python3.4/dist-packages (python 3.4)

jaredsalazar, 2016-06-14:
i experience this too. any solution for this?

LeonChambers, 2016-06-14:
So the problem is that python 3.5 somehow got installed alongside python 3.4 and pip3 is using the wrong version. I haven't encountered this personally, and it might not even be worth debugging because we're moving away from python 3 anyway. The current work is in integrating with ROS, which runs on Python 2. In particular, the python 3.5 requirement resulted from our use of pymata-aio, which we are replacing with the rosserial stack.

Mavni, 2016-06-14:
That's good to hear, as I spent hours on trying to install the Brain...
When is this new version due?
Should I wait for it or should I go ahead and deploy v1.0 for now?

Thanks

LeonChambers, 2016-06-15:
It depends on what your timeline and priorities are.

The first version of the software stack consists of microcontroller code in [gro-microcontroller](https://github.com/OpenAgInitiative/gro-microcontroller), some raspberry pi scripts in [gro-controller](https://github.com/OpenAgInitiative/gro-controller), an API in [gro-api](https://github.com/OpenAgInitiative/gro-api), and a UI in [gro-ui](https://github.com/OpenAgInitiative/gro-ui). That software stack works but is no longer under development.

All of the repos prefixed with `openag_` represent the new version of the software after a pretty major overhaul of the stack. For this repository in particular, the `ros` branch is what we currently use in house (and I'll probably soon merge the changes from it into master), but it's not finalized and documentation is a bit sparse. It runs with firmware code from [openag_phalange](https://github.com/OpenAgInitiative/openag_phalange) and the UI from [openag_ui](https://github.com/OpenAgInitiative/openag_ui). This repo is on the order of weeks away from being ready, but I'm not really sure how close the UI is.

closed 2016-06-20 by LeonChambers