OpenAgricultureFoundation/openag_brain #25: openag_ brain connection to arduino error

MisterBurnley, 2016-06-28:
Have been experimenting following install instructions for openag_brain. I have used the openag_brain_install_rpi  script and can't connect to my arduino. Googling the error suggests that it could be a baud rate mismatch. USB cable is OK as I can connect to the Arduino using Arduino IDE. Following the thought that it's something to do with the Baud rate I looked tried to find where it is set in the code.  The only setting I could find was in the rosserial_arduino_libs for the PlatformIo library which is 57600 don't know where to check if that is the same for rosserial,  Output from my raspberry pi 2 below. Any ideas ??

pi@raspberrypi:~/catkin_ws $ dir
build  CMakeLists.txt  devel  install  src
pi@raspberrypi:~/catkin_ws $ source ~/catkin_ws/devel/setup.bash
pi@raspberrypi:~/catkin_ws $ rosrun openag_brain main
Initializing the database
Generating launch file
... logging to /home/pi/.ros/log/3a642720-3d40-11e6-8d4e-b827eb391475/roslaunch-raspberrypi-4652.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:39862/

SUMMARY
========

PARAMETERS
 * /database: http://localhost:...
 * /development: False
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    api (openag_brain/api.py)
    handle_arduino (openag_brain/handle_arduino.py)
    topic_connector (openag_brain/topic_connector.py)

auto-starting new master
process[master]: started with pid [4664]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3a642720-3d40-11e6-8d4e-b827eb391475
process[rosout-1]: started with pid [4678]
started core service [/rosout]
process[api-2]: started with pid [4688]
process[topic_connector-3]: started with pid [4697]
process[handle_arduino-4]: started with pid [4698]
[ERROR] [WallTime: 1467125679.304795] Failed to update Arduino: [Errno 13] Permission denied
[ERROR] [WallTime: 1467125698.012291] Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

LeonChambers, 2016-06-28:
Looks like the `handle_arduino` module wasn't able to flash the arduino with the rosserial code. What happens if you try `rosrun openag_brain flash_arduino`? You could also try `sudo rosrun openag_brain flash_arudino` to see if that makes a difference. I suspect it might

MisterBurnley, 2016-06-28:
Hmm, must have something not set right !! here is the output, 
pi@raspberrypi:~ $ cd catkin_ws
pi@raspberrypi:~/catkin_ws $ source ~/catkin_ws/devel/setup.bash
pi@raspberrypi:~/catkin_ws $ rosrun openag_brain flash_arduino
/home/pi/catkin_ws/src/openag_brain/scripts/flash_arduino: line 2: pio: command not found
pi@raspberrypi:~/catkin_ws $ rosrun openag_brain flash_arduino
/home/pi/catkin_ws/src/openag_brain/scripts/flash_arduino: line 2: pio: command not found
pi@raspberrypi:~/catkin_ws $ sudo rosrun openag_brain flash_arduino
sudo: rosrun: command not found
pi@raspberrypi:~/catkin_ws $ dir
build  CMakeLists.txt  devel  install  src
pi@raspberrypi:~/catkin_ws $ cd src
pi@raspberrypi:~/catkin_ws/src $ dir
CMakeLists.txt	openag_brain
pi@raspberrypi:~/catkin_ws/src $ cd openag_brain
pi@raspberrypi:~/catkin_ws/src/openag_brain $ dir
CMakeLists.txt	Dockerfile	package.xml  scripts   srv
couchdb.ini	LICENSE		README.md    setup.py  test
doc		modules.launch	rosdoc.yaml  src
pi@raspberrypi:~/catkin_ws/src/openag_brain $ cd scripts
pi@raspberrypi:~/catkin_ws/src/openag_brain/scripts $ dir
flash_arduino	   init_db	load_fixture  software_modules
generate_firmware  install_pio	main	      update_launch
pi@raspberrypi:~/catkin_ws/src/openag_brain/scripts $ rosrun openag_brain flash_arduino
/home/pi/catkin_ws/src/openag_brain/scripts/flash_arduino: line 2: pio: command not found
pi@raspberrypi:~/catkin_ws/src/openag_brain/scripts $

LeonChambers, 2016-06-28:
Ok, so it looks like `/usr/local/bin` isn't in your `PATH`, so the terminal is having trouble finding the `platformio` executable. Run `export PATH=$PATH:/usr/local/bin` and try again. And I guess `sudo` won't actually work because of the environment being weird

MisterBurnley, 2016-06-28:
pi@raspberrypi:~/catkin_ws $ source ~/catkin_ws/devel/setup.bash
pi@raspberrypi:~/catkin_ws $ rosrun openag_brain flash_arduino
/home/pi/catkin_ws/src/openag_brain/scripts/flash_arduino: line 2: pio: command not found
pi@raspberrypi:~/catkin_ws $ sudo export PATH=$PATH:/usr/local/bin
sudo: export: command not found
pi@raspberrypi:~/catkin_ws $ export PATH=$PATH:/usr/local/bin
pi@raspberrypi:~/catkin_ws $ rosrun openag_brain flash_arduino
/home/pi/catkin_ws/src/openag_brain/scripts/flash_arduino: line 2: pio: command not found
Also set the PATH from another terminal, don't want to take up your valuable coding time so suggest that I reflash the card and go through the setup again and see if that works.

MisterBurnley, 2016-06-28:
Re-flashed the card and successfully ran, rosrun openag_brain main, got roslaunch server, ros_master_url and started core service. 
Opened a new terminal and couldn't flash the arduino as there wasn't an,  .openag/build directory. Used,  rosrun openag_brain generate_firmware,  which also built the directory.
Running rosrun openag_brain flash_arduino caused the system to download atmelavr.
A warning was generated,
looking for upload port/disk ... please install '99-platformio-udev.rules' and check that your boards PID and VID are listed in the rules.
then,
 scons:    [upload] Explicit Exit, status error please specify 'upload_port' for environment or use global, --upload-port , option.
Guess platformio needs the usb port address, not sure what it is or how to give it to the program unless I just put, --upload-port (address) on the end of, rosrun openag_brain flash_arduino ??

LeonChambers, 2016-06-29:
Ok. Kinda weird that re-flashing fixed things, but whatever.

If you haven't already, try unplugging the usb cable connected to the arduino and plugging it back in. If that doesn't work, try rebooting the Pi.

MisterBurnley, 2016-06-29:
Decided to install Arduino IDE and check that it could flash the mega and maybe set the speed and the port through the serial monitor settings. It was using /dev/ttyACM0 at 9600. The blink pin 13 code worked so I ran the flash_arduino script and it worked. The info generated gave a system wide settings address as /home/pi/.platformio/packages/tool-avrdude/avrdude.conf and a User Config address of /home/pi .avrduderc, it is on my system when I checked and the script skipped it. The port was set to /dev/ttyACM0. The read out for the baud rate was, overidding baud rate : 115200 which I took to mean that it was using 9600, which was the value in the Arduino IDE and not using 115200.

MisterBurnley, 2016-06-29:
Have got the ui running after a bit of fiddling with node and npm they needed updating from those in the updated and upgraded pi image. Used adafruit repo for node and, npm install raspberry pi, to get the latest version of npm.
The openag_ui built but didn't create a ./dist folder so used the html file in the dist folder in openag_ui. which also contains the main.js scripts and the assets folder which worked.
Had forgotten to use the -f default flag when running openag_brain have done that and everything seems to be working on the raspberry pi 2 now. Looking forward to further developments of the 
' Brain ' , any thoughts on adding an, AI ?

closed 2016-07-01 by MisterBurnley