OpenAgricultureFoundation/openag_brain #27: Using a PS3 webcam

MisterBurnley, 2016-07-13:
Have the software running and sending images to the couchdb from the PS3 webcam but had to edit the database config by hand as the program kept closing the webcam.
The log error was camera_info .yaml file , config file not found. The location of the file was the, .ros, directory in home.
I created the diectory and also the, top.yaml file, ( not sure on syntax for a .yaml file ) but it still didn't work. Looking through the code I think it is supposed to check the camera and update the database automatically. 
Have I missed an update command somewhere ?

LeonChambers, 2016-07-19:
In our working setup in the lab, it throws a warning about the missing .yaml file and runs anyway, so that's likely not your problem. You can check if there's a ROS topic named `/environment_1/camera_1/image_raw` and listen to it to see if images are being collected. You should also make sure that the ROS usb_cam package works with the particular webcam you have. We've only tested it with the one specific camera model that we've been using for testing, and it definitely doesn't work with all webcams.

MisterBurnley, 2016-07-19:
Thanks Leon, I have looked at the environmental_data_point in the couch database and the images from the PS3 webcam are stored there, so it must be working with the changes I made to software_module camera_1 in the database config, 
Sharpness 22,
camera_name 'top'
framerate 1
image_width 640
io_method 'nmap'
 image_height 480
camera_frame_id 'top'
autofocus false
pixel_format 'yuyv'
Now I leave out  the flags when runnimg, rosrun openag_brain main, so as not to update the database with the standard webcam settings.
That's great news, as the PS3 webcam can be picked up in the UK at games stores for  £2.00. Raspberry Pi camera module £18.99 +p&p on amazon. 5 for the price of one !!!!! the PS3 has 4 microphones as well. what could they be used for ????

closed 2016-07-21 by LeonChambers