OpenAgricultureFoundation/openag_brain #37: light_illuminance variable is missing in openag.var_types

serein7, 2016-09-04:
I'm experimenting with writing a [driver](https://github.com/serein7/openag_bh1750) for cheap light sensor based on Bh1750 chip . All seams to be working fine and I even see lux reading in [OpenAg UI](http://joxi.ru/bmovYEZhMKOedr).

Just minor issue I've found looking in logs:

> 2016-09-04T07:50:49.260454659Z [WARN] [WallTime: 1472975449.259701] Cloning into 'openag_bh1750'...
>**2016-09-04T07:50:49.619175349Z [WARN] [WallTime: 1472975449.618340] Encountered a module output that references a non-existant variable: Output "light_illuminance" of module "bh1750_1"**
>2016-09-04T07:50:51.696040275Z [WARN] [WallTime: 1472975451.695208] Cloning into 'openag_firmware_module'...
>2016-09-04T07:50:54.135100866Z [WARN] [WallTime: 1472975454.134453] Cloning into 'openag_am2315'...
>2016-09-04T07:51:23.426972246Z [WARN] [WallTime: 1472975483.426506] Copyright (c) 2000-2005 

Looking at code I suspect that light_illuminance is missing in EnvVr (openag.var_types)

   
    valid_vars = list(EnvVar.items.keys())
    for module_id, module_info in modules.items():
        for output_name, output_info in module_info["outputs"].items():
            if not output_info["variable"] in valid_vars:
                rospy.logwarn(
                    "Encountered a module output that references a "
                    'non-existant variable: Output "%s" of module "%s"',
                    output_name, module_id
                )
                continue

gordonbrander, 2016-09-12:
Yeah this seems to be the case. @LeonChambers do you have a preference for what this variable should be? `light_illuminance`?

LeonChambers, 2016-09-14:
Yea, `light_illuminance` is fine

gordonbrander, 2016-09-15:
Fixed by https://github.com/OpenAgInitiative/openag_python/pull/3.

closed 2016-09-15 by gordonbrander