OpenAgricultureFoundation/openag_brain #40: Error running firmware created for TSL2561

AdrianLLu, 2016-09-13:
@gordonbrander 

What I did so far is install the openag_brain on my raspberry pi, test the tsl2561 code I found on an Arduino, and attempt to wrap the tsl2561 code as instructed in the doc files on the OpenAg github. My github for the tsl2561 files here.

The first error I received appears to be related to the Arduino code:
`In file included from /home/pi/Documents/branch_tsl2561/_build/src/src.ino:1:0:
.pioenvs/megaatmega2560/module/branch_tsl2561.h:128:30: error: cannot declare variable ‘TSL2561’ to be of abstract type ‘TSL2561_CalculateLux’
extern TSL2561_CalculateLux  TSL2561;
^`

Here's the full error log:
```
ipi@raspberrypi:~/Documents/branch_tsl2561` $ openag firmware run_module -p csv -t upload
OpenAg firmware project initialized!
Parsing firmware module type "module" from lib folder
Parsing firmware module "module" from modules file
Installing library [ 345 ]:
Already installed
Updating "openag_firmware_module"
Already up-to-date.
[Mon Sep 12 06:28:08 2016] Processing megaatmega2560 (platform: atmelavr, board: megaatmega2560, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/megaatmega2560/src/tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DOPENAG_CATEGORY_SENSORS -DOPENAG_CATEGORY_ACTUATORS -DARDUINO=10608 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/openag_firmware_module -I.pioenvs/megaatmega2560/rosserial_arduino_ID345 -I.pioenvs/megaatmega2560/module -Isrc src/tmp_ino_to.cpp
In file included from /home/pi/Documents/branch_tsl2561/_build/src/src.ino:1:0:
.pioenvs/megaatmega2560/module/branch_tsl2561.h:128:30: error: cannot declare variable ‘TSL2561’ to be of abstract type ‘TSL2561_CalculateLux’
extern TSL2561_CalculateLux  TSL2561;
^
.pioenvs/megaatmega2560/module/branch_tsl2561.h:106:7: note:   because the following virtual functions are pure within ‘TSL2561_CalculateLux’:
class TSL2561_CalculateLux : public Module {
^
In file included from .pioenvs/megaatmega2560/module/branch_tsl2561.h:36:0,
from /home/pi/Documents/branch_tsl2561/_build/src/src.ino:1:
.pioenvs/megaatmega2560/openag_firmware_module/openag_module.h:18:18: note: 	virtual void Module::begin()
virtual void begin() = 0;
^
.pioenvs/megaatmega2560/openag_firmware_module/openag_module.h:19:18: note: 	virtual void Module::update()
virtual void update() = 0;
^
/home/pi/Documents/branch_tsl2561/_build/src/src.ino:13:1: error: ‘Tsl2561’ does not name a type
std_msgs::Float32 module_lux_msg;
^
/home/pi/Documents/branch_tsl2561/_build/src/src.ino: In function ‘void setup()’:
/home/pi/Documents/branch_tsl2561/_build/src/src.ino:20:3: error: ‘module’ was not declared in this scope
}
^
/home/pi/Documents/branch_tsl2561/_build/src/src.ino: In function ‘void loop()’:
/home/pi/Documents/branch_tsl2561/_build/src/src.ino:29:3: error: ‘module’ was not declared in this scope
if (module.get_lux(module_lux_msg)) {
^
avr-gcc -o .pioenvs/megaatmega2560/FrameworkArduino/wiring_shift.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DOPENAG_CATEGORY_SENSORS -DOPENAG_CATEGORY_ACTUATORS -DARDUINO=10608 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant .pioenvs/megaatmega2560/FrameworkArduino/wiring_shift.c
avr-ar rcs .pioenvs/megaatmega2560/libopenag_firmware_module.a
avr-ranlib .pioenvs/megaatmega2560/libopenag_firmware_module.a
avr-g++ -o .pioenvs/megaatmega2560/rosserial_arduino_ID345/duration.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega2560 -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_MEGA2560 -DOPENAG_CATEGORY_SENSORS -DOPENAG_CATEGORY_ACTUATORS -DARDUINO=10608 -I.pioenvs/megaatmega2560/FrameworkArduino -I.pioenvs/megaatmega2560/FrameworkArduinoVariant -I.pioenvs/megaatmega2560/openag_firmware_module -I.pioenvs/megaatmega2560/rosserial_arduino_ID345 -I.pioenvs/megaatmega2560/module .pioenvs/megaatmega2560/rosserial_arduino_ID345/duration.cpp
scons: *** [.pioenvs/megaatmega2560/src/tmp_ino_to.o] Error 1
==================================================== [ ERROR ] Took 5.97 seconds ====================================================
Error: Compilation failed
```

Let me know if you need more information.

Cheers!
-A

LeonChambers, 2016-09-14:
Well, the code for this is actually in the `openag_python` repo, not this one.

But anyway, it looks like you haven't defined `begin` and `update` functions for your `Module` subclass. There might be other problems too, but I can't tell without looking at your module code.

AdrianLLu, 2016-09-16:
Gotcha @LeonChambers . Shall I move the issue there?

Here's the link to my [code](https://github.com/branchworks/branch_tsl2561).

Thanks for the help! Let me know if you need anything else from me.

LeonChambers, 2016-09-24:
The `class_name` in your `module.json` file doesn't match the name of the class you wrote, and the class doesn't inherit from the `Module` class from openag_firmware_module. If you fix both of those things, the code should compile

AdrianLLu, 2016-10-03:
Thanks for the reply @LeonChambers . Hm, not being a programmer nor fully understanding the brain architecture, I think I'm going to have to take a step back and learn some C++ and spend time on the documentation projects  before I try to tackle this again.

closed 2017-01-10 by LeonChambers