OpenAgricultureFoundation/openag_brain #41: Can not add PID software_module

serein7, 2016-09-14:
@LeonChambers, I'm trying to setup feedback loop on light_illuminance. Based on my understanding of PID fixture in software_module_types the only required parameter is variable. So I add following into software_module:

`{
   "_id": "light_pid",
   "type": "openag_brain:pid.py",
   "parameters": {
       "variable": "light_illuminance"
   }
}`

My assumption that, nevertheless I did not specify other params like Kd, Ki, Kp and etc this module should still be wired correctly and run, even if PID will not produce valid outputs. 
However, I see following error in logs:

>2016-09-14T19:18:33.367723613Z Processing module "light_pid" from server
2016-09-14T19:18:33.366726437Z Traceback (most recent call last):
2016-09-14T19:18:33.367859705Z   File "/home/pi/catkin_ws/src/openag_brain/scripts/main", line 117, in <module>
2016-09-14T19:18:33.367948610Z     fixtures=vals.fixtures, screen=vals.screen
2016-09-14T19:18:33.368021057Z   File "/home/pi/catkin_ws/src/openag_brain/scripts/main", line 58, in main
2016-09-14T19:18:33.368152306Z     commands.update_launch(server)
2016-09-14T19:18:33.368564228Z   File "/home/pi/catkin_ws/src/openag_brain/src/openag_brain/commands/update_launch.py", line 99, in update_launch
2016-09-14T19:18:33.368667091Z     module_type = SoftwareModuleType(module_types_db[module["type"]])
2016-09-14T19:18:33.371513256Z   File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 192, in __call__
2016-09-14T19:18:33.371671327Z     return self._compiled([], data)
2016-09-14T19:18:33.371751066Z   File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 486, in validate_dict
2016-09-14T19:18:33.371827107Z     return base_validate(path, iteritems(data), out)
2016-09-14T19:18:33.371911272Z   File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 324, in validate_mapping
2016-09-14T19:18:33.371987522Z     raise er.MultipleInvalid(errors)
2016-09-14T19:18:33.372056740Z voluptuous.error.MultipleInvalid: not a valid value for dictionary value @ data[u'parameters'][u'Kd'][u'type']

Any idea?

LeonChambers, 2016-09-15:
Yea, this is a bug in the `default.json` fixture. There are several parameters of type "double" for the PID module. These should have type "float" instead. You can fix it locally by modifying the entry in `software_module_type` in your database for the PID module according. The fixture should be fixed too though so other people don't encounter this

serein7, 2016-09-15:
Ok, I see that it must be `float` based on openag [schema definishion](https://github.com/OpenAgInitiative/openag_python/blob/db2e55b9bd07c0926028b653b65d61cace8bc912/openag/models.py#L378), but it looks like roslaunch does not know this type... I guess some kind of mapping is missing...

serein7, 2016-09-15:
Further studying the code I see something like mapping at https://github.com/OpenAgInitiative/openag_brain/blob/e7694bd2e08f9451005c08bf756d8a164e70e10b/src/openag_brain/commands/update_launch.py#L131
However, could not find any sample. I guess I need to add to PID type definition something like
`"mappings": {"from":"float", "to":"std_msgs/Float64" }` ? or just wait until you get it fixed :-)

LeonChambers, 2016-09-15:
Oh, mappings don't actually exist anymore in the object model; that script is just out of date. I think the right approach for this is to change the `update_launch.py` script to map from "float" to "double" automatically so the roslaunch file is actually valid.

serein7, 2016-09-15:
Well, here is the [PR](https://github.com/OpenAgInitiative/openag_brain/pull/42). Please mind, however, that while I was able to fix the error with this patch, I'm not very confident neither with Python, neither with ROS.

LeonChambers, 2016-09-19:
Fixed in #42

closed 2016-09-19 by LeonChambers