OpenAgricultureFoundation/openag_brain #43: Can't build docker image

serein7, 2016-09-17:
When I try to follow [instructrions](https://github.com/OpenAgInitiative/openag_brain/blob/cafacbb936617fa087f3a2fd85f3265de90fdcf7/doc/docker.rst#compiling-and-publishing-a-new-docker-image) with `docker build -t openag/rpi_brain`
first I get an error like this:
>docker: "build" requires 1 argument.
>See 'docker build --help'.
>Usage:	docker build [OPTIONS] PATH | URL | -
>Build an image from a Dockerfile

Well, I assume that name is missing, so I make another time with `docker build -t openagbraindockerrpi_brain openag/rpi_brain`. But now it says:
>unable to prepare context: unable to evaluate symlinks in context path: lstat /home/serein/Projects/openAgInitiative/openag_brain/openag/rpi_brain: no such file or directory

Indeed there is not such directory or file in git repo.

LeonChambers, 2016-09-17:
The missing argument is the path to the folder containing the Dockerfile. If you run the command from the openag_brain folder, it should be `docker build -t openag/rpi_brain .`. Docs should be updated accordingly.

serein7, 2016-09-17:
@LeonChambers, thank you. One more question from docker newbe - am I correct to assume that image build needs to be performed on rpi?

LeonChambers, 2016-09-17:
Yup. Technically any machine with an ARM CPU would work, but I've been doing them on the Pi

closed 2016-09-19 by LeonChambers