OpenAgricultureFoundation/openag_brain #45: How to wire PID with sensors and actuators?

serein7, 2016-09-17:
Originally posted question [here](http://forum.openag.media.mit.edu/t/openag-ui-web-app-build-diary/83/36?u=serein), but decided that github is better place to discuss

Here is my setup:
1. light_illuminance sensor, which is represented by following ROS topics:
`/sensors/bh1750_1/light_illuminance/filtered
/sensors/bh1750_1/light_illuminance/raw
/environment_1/measured/light_illuminance`
2. binary actuator connected to LED lights. Which is represented by ROS topic `/actuators/light_actuator_1/cmd` and I can manually turn lights On and Off by sending True or False with `rostopic pub` command.
3. simple recipe with light_illuminance set to 1500 lux. I see that this recipe creates ROS topic `/environment_1/desired/light_illuminance` and when I run `rostopic echo /environment_1/desired/light_illuminance` I see `data: 1500` being constantly published.
4. light_pid controller with variable parameter set to `light_illuminance`. This in turn creates following topics:
`/light_illuminance
/light_illuminance_cmd
/light_illuminance_desired
/light_illuminance_down_cmd
/light_illuminance_up_cmd`

Now, what I don't get is how to connect a) `/environment_1/measured/light_illuminance` output to PID's input `/light_illuminance` , b) `environment_1/desired/light_illuminance` output to PID's `/light_illuminance_desired` input and how to c) translate `/light_illuminance_cmd` with numerical output to actuator's input `/actuators/light_actuator_1/cmd`, which expects boolean.

I checked `topic_connector.py` software module, but looks like it is doing a global mapping from low-level `/sensors/*` topics to `/environment_1/*`. 

I hope you can point me to the source code or sample on how to wire PID with sensors and actuators.

LeonChambers, 2016-09-17:
The PID module is a work in progress, so there are no usage examples. If you can get it working, that's great, but I can't really help you at this point.

serein7, 2016-09-17:
Thanks anyway. It still helps to understand that something is just not finished, rather than rage at yourself that you can't get it work :-)

MisterBurnley, 2016-09-17:
Great work Serein, the above description of how the data flows from one set of apps to another is most illuminating, it  gives a ' helicopter ' view of the software so that each  parts role in the whole picture is clearer. Guess the team is pretty busy at the moment on their specific tasks with the November deadline looming. Exciting times !!!

serein7, 2016-09-23:
@LeonChambers
>The PID module is a work in progress, so there are no usage examples. If you can get it working, that's great, but I can't really help you at this point.

Well, I've got it working. 

I had to create a new module `pid_connector` which connects appropriate topics from environment's sensors and actuator to the namespace PID is expecting.

The one thing brought my attention during this exercise is that currently actuators logical namespace is not linked to an environment. I.e., unlike sensors, which are connected from `/sensors/bh1750_1/light_illuminance`  to `/environment_1/measured/light_illuminance`, the actuator's topic stays standalone like this `/actuators/light_actuator_1/cmd`. Looks a bit out of order to me. Maybe I miss something?

gordonbrander, 2016-10-18:
This discrepancy has been fixed.

closed 2017-01-25 by LeonChambers