OpenAgricultureFoundation/openag_brain #52: start_recipe API is broken

serein7, 2016-09-23:
The latest commit which has it working is 88b01ee
Than starting from dc25676cd1d34c87b635ca050e89d2f2ddbf7d4f whenever I try to call recipe_start I get following exception:

>Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gevent/pywsgi.py", line 508, in handle_one_response
    self.run_application()
  File "/usr/lib/python2.7/dist-packages/gevent/pywsgi.py", line 494, in run_application
    self.result = self.application(self.environ, self.start_response)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1836, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1820, in wsgi_app
    response = self.make_response(self.handle_exception(e))
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1403, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python2.7/dist-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/catkin_ws/src/openag_brain/src/openag_brain/software_modules/api.py", line 127, in perform_service_call
    res = rosservice.call_service(service_name, [args])[1]
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rosservice/__init__.py", line 411, in call_service
    rospy.init_node('rosservice', anonymous=True)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 271, in init_node
    raise rospy.exceptions.ROSException("rospy.init_node() has already been called with different arguments: "+str(_init_node_args))
ROSException: rospy.init_node() has already been called with different arguments: ('api', ['/home/pi/catkin_ws/src/openag_brain/src/openag_brain/software_modules/api.py', '__name:=api', '__log:=/home/pi/.ros/log/7a3f92f4-815a-11e6-baa8-b827eba03916/api-5.log'], False, None, False, False)

@gordonbrander, it seems that something in dc25676cd1d34c87b635ca050e89d2f2ddbf7d4f broke it.

gordonbrander, 2016-10-12:
Blerg! Looking into it, thanks!

closed 2016-10-17 by gordonbrander

gordonbrander, 2016-10-17:
@serein7 thanks for your help fixing this!