OpenAgricultureFoundation/openag_brain #56: Running without locale set results in mysterious looping error

gordonbrander, 2016-10-03:


LeonChambers, 2017-01-10:
Uh... Not sure what you mean by this. Any idea how I could reproduce it?

gordonbrander, 2017-01-25:
Yeah, install Raspbian Jesse Lite (I think it comes without a locale by default) and run openag_brain_install_rpi. When you try running `rosrun openag_brain main` you'll see what I mean.

gordonbrander, 2017-01-27:
Just hit this on another Raspberry Pi [image]

wsnook, 2017-01-29:
@gordonbrander  According to pages 137-138 of [Raspberry Pi Hacks: Tips & Tools for Making Things with the Inexpensive Linux Computer](https://books.google.com/books?id=pdWQAwAAQBAJ&pg=PA137&lpg=PA137&dq=ros+raspberry+pi+locale&source=bl&ots=8qDq3Org_V&sig=uQzX2sRBPBJeIVDVCtjwEb_L7_A&hl=en&sa=X&ved=0ahUKEwivgL7ev-bRAhVLxYMKHeqGBCoQ6AEIPzAF#v=onepage&q=ros%20raspberry%20pi%20locale&f=false) (link to google books preview), that error happens because code in ROS specifically requests locales like `en_US.UTF-8` and others that aren't `en_GB`. I was just looking at locale settings in `raspi-config` tonight on a fresh January 2017 Jessie Lite install--it only had `en_GB.UTF-8` configured.

Also, the [Ubuntu ARM ROS Indigo wiki docs](http://wiki.ros.org/indigo/Installation/UbuntuARM#Set_your_Locale) say "Boost and some of the ROS tools require that the system locale be set."

This is on my radar because I'm in the process of attempting to install Jessie Lite followed by a global brain install, and I'm writing up my procedure as I go:  https://gist.github.com/wsnook/24ea657cf600a3802b262a2798f8894b.

I haven't hit your error yet. But, out of habit, I already included instructions about setting locale, wifi country, and timezone. My intention is to post about my gist on the OpenAg forum once I get the brain and UI running. I'd also be happy to make a documentation pull request (how to set locale) against one of the READMEs or maybe sphinx docs if that would be helpful.