OpenAgricultureFoundation/openag_brain #58: handle_arduino fixture arguments are never used

gordonbrander, 2016-10-12:
handle_arduino.py is intended to have a serial port argument ([see docblock](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/handle_arduino.py#L12))

```
The module reads from a command line argument `serial_port` which is the UNIX
path to the serial port to which the Arduino is connected (e.g. "/dev/ttyACM0")
```

And the [arduino_handler fixture is configured with arguments](https://github.com/OpenAgInitiative/openag_brain/blob/master/fixtures/default.json#L398), however, these arguments are never used. [The call to open the serial port is hardcoded](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/handle_arduino.py#L67).

pratapmn, 2016-10-16:
Just for everyones info. Some older arduino chips connect to /dev/ttyUSB0 but openag handle arduino code is hardcoded to read from ttyACM0. I dug through the code to find and replace ttyACM0 to ttyUSB0 to make it not fail at opening port.

Locally in my openag build I updated below two files
https://github.com/OpenAgInitiative/openag_brain/blob/master/fixtures/default.json#L402
https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/handle_arduino.py#L78

gordonbrander, 2016-10-21:
Maybe related: my Arduino is plugged in, but am now getting: `Error opening serial: could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'`.

LeonChambers, 2017-01-10:
Clossed in #96

closed 2017-01-10 by LeonChambers