OpenAgricultureFoundation/openag_brain #70: direct_controller.py can only handle Float64 type messages

gordonbrander, 2016-10-21:
`direct_controller.py` [can only work with Float64 messages](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/direct_controller.py#L18).

We should have an analog to `direct_controller.py` that works with binary values so we can use it with `binary_actuator.py`.

gordonbrander, 2016-10-21:
Update: same problem with topic connector. [src_topic_type is hardcoded as Float64](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/topic_connector.py#L66).

LeonChambers, 2016-10-23:
Just to make explicit some relevant design decisions, right now, everything under `/environment` is a Float64, but things under `/sensors` and `/actuators` are allowed to have different types reflecting the actual precision of the hardware as long as they can be easily converted to floats. I think supporting Bools in `/environment` could make sense, but disallowing all other types makes the code simpler so that, for example, a PID module that operates on floats can be used to control any piece of hardware regardless of precision. I think originally my plan was to just encode Bools as Floats (0.0 for False and 1.0 for True) to make the code simpler, but having them actually typed as booleans is definitely more clear.

tl;dr let's make topic_connector support Bools

Spaghet, 2017-05-13:
http://wiki.ros.org/std_msgs

>Therefore, while the messages in this package can be useful for quick prototyping, they are NOT intended for "long-term" usage...we recommend that existing messages be used, or new messages created, that provide meaningful field name(s). 

Interesting point from ROS wiki