OpenAgricultureFoundation/openag_brain #72: measured and desired topics are not prefixed by environment

gordonbrander, 2016-10-21:
Is this intentional?

[image]

(I've configured binary_actuator with `"input": {"variable": "light_illuminance"}`)

gordonbrander, 2016-10-21:
Update: never mind. No collision, because the topic is `/actuators/<id>/cmd`.

LeonChambers, 2016-10-23:
My guess is that you didn't specify an environment for the controller. We should add a check to the controller code that prevents this from happening. For reference, I did a check like this for the video_writer module [here](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/video_writer.py#L153)

closed 2016-10-26 by gordonbrander