OpenAgricultureFoundation/openag_brain #78: Remove dependency on ROS for building docs

gordonbrander, 2016-11-04:
The one annoying thing to figure out is how to load up the version number from the XML file without pulling in all of ROS.

serein7, 2016-11-04:
Download single file from other repo and parse it out? http://stackoverflow.com/questions/4604663/download-single-files-from-github

gordonbrander, 2016-11-04:
I think we can keep the file in this repo and use the `xml` package to grab the version number.

LeonChambers, 2017-01-10:
Well, if we let the [ROS buildfarm](http://wiki.ros.org/build.ros.org) compile our docs, then this isn't really a problem, and ROS is willing to host documentation for open source projects for free. However, we might want all of our documentation hosted in the same place, and not all of our code is in ROS packages.

That being said, parsing the XML file ourselves really isn't that hard, but we should figure out where to host the docs anyway.

rallytime, 2017-03-07:
I think this issue can be closed now? :)

closed 2017-03-07 by gordonbrander