OpenAgricultureFoundation/openag_brain #79: Pause / scrub / start recipe @ time

gordonbrander, 2016-11-08:
We should create the ability to pause a recipe (stop and re-start at a given time).

- When restarting, we should make sure not to emit another recipe_start datapoint. We want datapoints to be associated with current recipe. Instead, we should emit pause/unpause datapoints.
- We should also be able to start a recipe at a given time (for troubleshooting, etc).

For recipes formats without a definite end state, this is a bit more complicated.

- The recipe can ignore the start time if it doesn't make sense.
- For pause/unpause, the recipe should resume from last state in a way that makes sense (it may try to extrapolate the interim).

gordonbrander, 2016-12-19:
I think this should be scoped down to "start recipe at time". Recipes can already pick back up after the Pi is rebooted.

LeonChambers, 2016-12-19:
Agreed. Being able to start a recipe at a given time is actually very easy. The recipe_handler already does this internally on startup if it sees that a recipe should be in progress. You just have to add another parameter to the StartRecipe service definition to expose this functionality to the rest of the system.

Spaghet, 2017-03-06:
Heavily agree with this!

I have growlights that keep me awake at night...