OpenAgricultureFoundation/openag_brain #80: Unusual Serial port error when reading v1 sensors

MisterBurnley, 2016-11-10:
Have had multiple errors compiling ROS on the latest version of rasbian so decide to use the docker image to see if TSL2561 sensor code worked. Discussed on the forum hackers needed V1 sensors on V2 thread with Gordon, Serein and Khallad. Had a working system using just the dht22 and TSL2561 sensors last night.
Trying it again today ( Nov 10th ) it initially didn't work. When I shelled into the containers the topics were there but no data was displayed when I used rostopic echo. I remembered trying, platformio serialports monitor, yesterday, from a second terminal to see if there was any data being sent on the serial port. Trying it again today gave the following error,  

pi@raspberrypi:~ $ platformio serialports monitor
--- Miniterm on /dev/ttyACM0  9600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�␀�
␀␀␀␀␀␀␀␀␀���]␀�␀␀}␀␌␀␀␀/diagnostics␟␀␀␀diagnostic_sgs/Diagnoticra ␀␀
--- exit ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/pio_env/local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 452, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/home/pi/pio_env/local/lib/python2.7/site-packages/serial/serialposix.py", line 468, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

I then went back to the first terminal and issued the rostopic echo command and it started sending data to the terminal screen, 

pi@1a5c21b7d67f:~$ rostopic echo /environment_1/measured/air_humidity
data: 81.5330047607
---
data: 81.5231018066
---
data: 81.5161743164
---
data: 81.4513244629
---
data: 81.4359283447
---
data: 81.4251480103

When I started the hosted UI the data was traced on the graph and updated. It's been working for approx 2 hours so far.. I'm happy with getting the sensor code to work and raised this issue as I'm not sure why it happens ? I was expecting everything to launch when the docker image was started ?

LeonChambers, 2016-11-10:
Well, the docker image starts a process to listen on the serial port, so running `pio serialports monitor` alongside it would confuse the OS, which is why you get the error about "multiple access on port". I'm not sure why the readings didn't show up for you at first, but running `pio serialports monitor` probably refreshed the serial connection between the arduino and the ROS node reading from it, which happened to fix it. In general, when you aren't seeing readings, you should also check the `/diagnostics` topic to see if there are any error messages. That would help us try to debug this. As is, there's really no way for me to replicate this, so there's not much I can do to help.

MisterBurnley, 2016-11-10:
Readout from /diagnostics , when I issued platformio serialports monitor  with the image running,

header: 
  seq: 43
  stamp: 
    secs: 1478801374
    nsecs: 234977960
  frame_id: ''
status: 
  - 
    level: 2
    name: rosserial_python
    message: Mismatched protocol version in packet: lost sync or rosserial_python is from different ros release than the rosserial client
    hardware_id: ''
    values: 
      - 
        key: last sync
        value: Thu Nov 10 18:09:29 2016
      - 
        key: last sync lost
        value: Thu Nov 10 15:04:21 2016
---
header: 
  seq: 44
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: tsl2561_1
    message: ''
    hardware_id: none
    values: []
  - 
    level: 0
    name: dht22_1
    message: ''
    hardware_id: none
    values: []

 This is the error I was getting when I was using my local copy of ROS and openag_brain, the reason I decide to re-install Ros.  

Here is the the diagnostic output after issuing Docker-compose stop and then re-starting with docker-compose start. 

pi@1a5c21b7d67f:~$ rostopic echo /diagnostics
header: 
  seq: 1
  stamp: 
    secs: 1478802485
    nsecs: 412194967
  frame_id: ''
status: 
  - 
    level: 2
    name: rosserial_python
    message: no sync with device
    hardware_id: ''
    values: 
      - 
        key: last sync
        value: Thu Nov 10 18:27:50 2016
      - 
        key: last sync lost
        value: Thu Nov 10 18:28:05 2016
---

and now after issuing plo serialports monitor from a second terminal,

header: 
  seq: 139
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: tsl2561_1
    message: ''
    hardware_id: none
    values: []
  - 
    level: 2
    name: dht22_1
    message: Failed to read from sensor
    hardware_id: none
    values: []
---
header: 
  seq: 140
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: tsl2561_1
    message: ''
    hardware_id: none
    values: []
  - 
    level: 0
    name: dht22_1
    message: ''
    hardware_id: none
    values: []
---
header: 
  seq: 141
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: tsl2561_1
    message: ''
    hardware_id: none
    values: []
  - 
    level: 0
    name: dht22_1
    message: ''
    hardware_id: none
    values: []
---
header: 
  seq: 142
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: tsl2561_1
    message: ''
    hardware_id: none
    values: []
  - 
    level: 0
    name: dht22_1
    message: ''
    hardware_id: none
    values: []
---
Error response from daemon: Container 1a5c21b7d67f1b12b152eb3bc224f27bb57600468586c8291c9c444d4df7e976 is restarting, wait until the container is running
pi@raspberrypi:~/openag_brain_docker_rpi $

LeonChambers, 2016-11-10:
Ok, so the only weird thing I see here is the "no sync with device message". I've often seen these messages right after starting the container, and they usually stop showing up if you wait a while, but that's definitely not how this should work. We're trying to figure out what exactly causes that and how to avoid it (or fix it when it happens).

MisterBurnley, 2016-11-11:
As you know I set myself the challenge to hack together a driver for the TSL2561 sensor. I have confirmation that the goal has been achieved with yours and the teams help and guidance. Also the code that Khallad has created is shown to work  So, as the saying goes, ' I'm a happy bunny ' 
I'll leave figuring out the causes to your team, although I find solving or even trying to solve software problems very addictive.

Spaghet, 2017-07-20:
I think this is obsolete now that we've overhauled the serial communication protocol.

closed 2017-07-20 by Spaghet