OpenAgricultureFoundation/openag_brain #101: ImportError: cannot import name LIGHT_ILLUMINANCE

aguadopd, 2017-01-17:
`sensor_persistence` software module is crashing after `ImportError: cannot import name LIGHT_ILLUMINANCE`. Without `sensor_persistence`, the sensor data is not saved into the database.

Tested on 2a1c64a7e3f1c5dc701c0910866cecad67b18b3d. Problem is gone in b454ff3a044875acc7dae2cb7109d7e92d5463c1, and maybe more recently.

Full console output for the default fixture:

```
iki@c3po:~/catkin_ws/src/openag_brain$ rosrun openag_brain main -f default
Initializing the database
Applying CouchDB configuration  [####################################]  100%
Creating databases  [####################################]  100%
Pushing design documents
Applying fixture /home/viki/catkin_ws/src/openag_brain/fixtures/default.json
environment  [####################################]  100%
software_module_type  [####################################]  100%
firmware_module_type  [####################################]  100%
software_module  [####################################]  100%
Generating launch file
Processing module "arduino_handler" from server
Processing module "topic_filter_1" from server
Processing module "recipe_handler_1" from server
Processing module "api" from server
Processing module "topic_connector" from server
Processing module "topic_filter" from server
Processing module "video_writer_1" from server
Processing module "image_persistence_1" from server
Processing module "sensor_persistence_1" from server
Spawning software modules
... logging to /home/viki/.ros/log/39030bb4-dcf3-11e6-985d-080027b46cdd/roslaunch-c3po-3688.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://c3po:47154/

SUMMARY
========

PARAMETERS
 * /arduino_handler/serial_port: /dev/ttyACM0
 * /arduino_handler/should_flash: False
 * /environments/environment_1/image_persistence_1/min_update_interval: 3600
 * /environments/environment_1/sensor_persistence_1/max_update_interval: 600
 * /environments/environment_1/sensor_persistence_1/min_update_interval: 5
 * /environments/environment_1/video_writer_1/timelapse_scaling_factor: 86400
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /environments/environment_1/
    image_persistence_1 (openag_brain/image_persistence.py)
    recipe_handler_1 (openag_brain/recipe_handler.py)
    sensor_persistence_1 (openag_brain/sensor_persistence.py)
    topic_filter_1 (openag_brain/topic_filter.py)
    video_writer_1 (openag_brain/video_writer.py)
  /
    api (openag_brain/api.py)
    arduino_handler (openag_brain/handle_arduino.py)
    topic_connector (openag_brain/topic_connector.py)
    topic_filter (openag_brain/topic_filter.py)

auto-starting new master
process[master]: started with pid [3700]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 39030bb4-dcf3-11e6-985d-080027b46cdd
process[rosout-1]: started with pid [3713]
started core service [/rosout]
process[arduino_handler-2]: started with pid [3724]
process[environments/environment_1/topic_filter_1-3]: started with pid [3731]
[... 171 lines cut ...]
[WARN] [WallTime: 1484685372.356441] PollIndex                     : 3
[WARN] [WallTime: 1484685372.357084] PollValue                     : 0x53
[WARN] [WallTime: 1484685372.357447] Memory Detail                 :
[WARN] [WallTime: 1484685372.357957] Block Poll               Page                       Polled
[WARN] [WallTime: 1484685372.358447] Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
[WARN] [WallTime: 1484685372.358995] ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
[WARN] [WallTime: 1484685372.359601] eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
[WARN] [WallTime: 1484685372.360872] flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
[WARN] [WallTime: 1484685372.361952] lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
[WARN] [WallTime: 1484685372.369115] hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
[WARN] [WallTime: 1484685372.372773] efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
[WARN] [WallTime: 1484685372.373843] lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
[WARN] [WallTime: 1484685372.374941] calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
[WARN] [WallTime: 1484685372.376206] signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
[WARN] [WallTime: 1484685372.376764] Programmer Type : Wiring
[WARN] [WallTime: 1484685372.377204] Description     : Wiring
[WARN] [WallTime: 1484685372.377641] Programmer Model: AVRISP
[WARN] [WallTime: 1484685372.378770] Hardware Version: 15
[WARN] [WallTime: 1484685372.387715] Firmware Version Master : 2.10
[WARN] [WallTime: 1484685372.390442] Vtarget         : 0.0 V
[WARN] [WallTime: 1484685372.394511] SCK period      : 7.6 us
[WARN] [WallTime: 1484685372.397500] Reading | ################################################## | 100% 0.02s
[WARN] [WallTime: 1484685374.870113] Writing | ################################################## | 100% 2.43s
[WARN] [WallTime: 1484685376.824805] Reading | ################################################## | 100% 1.95s
```

LeonChambers, 2017-01-17:
I recently ran into this problem because I was running with an outdated version of `openag_python`. Try updating `openag_python` from the [git repo](https://github.com/OpenAgInitiative/openag_python) and let me know if the problem persists.

aguadopd, 2017-01-17:
Well, pardon my ignorance, but I don't remember having installed it before; I've always been following the instructions found [here in the docs](https://github.com/OpenAgInitiative/openag_brain/blob/master/doc/getting_started.rst) (which are, by the way, more correct than the current main readme.md of the repo).

I cannot find it anywhere installed, so probably I've been missing that step. However I can see that the [openag_brain_install_rpi script](https://github.com/OpenAgInitiative/openag_brain_install_rpi/blob/master/install.sh) is installing it.

Should I, then, just do:
```
mkdir -p ~/openag/src
cd ~/openag/src
git clone https://github.com/OpenAgInitiative/openag_python.git
cd openag_python
sudo pip install -e .
```
?

PS: It seems I've been missing an important part of the system. What's the role of openag_python and how is it related to openag_brain? How could I work before without it?

aguadopd, 2017-01-17:
Just tried and now I get

```
setting /run_id to b6882eea-dcfb-11e6-918f-080027b46cdd
process[rosout-1]: started with pid [12474]
started core service [/rosout]
process[arduino_handler-2]: started with pid [12489]
process[environments/environment_1/recipe_handler_1-3]: started with pid [12492]
process[environments/environment_1/video_writer_1-4]: started with pid [12493]
process[api-5]: started with pid [12494]
process[topic_connector-6]: started with pid [12495]
process[topic_filter-7]: started with pid [12496]
process[image_persistence_1-8]: started with pid [12497]
process[sensor_persistence_1-9]: started with pid [12498]
[WARN] [WallTime: 1484688436.164413] Timelapse scaling factor not specified. Defaulting to 86400
[WARN] [WallTime: 1484688436.268509] Serial port for arduino_handler not specified. Defaulting to /dev/ttyACM0
Traceback (most recent call last):
  File "/home/viki/catkin_ws/src/openag_brain/src/openag_brain/software_modules/image_persistence.py", line 88, in <module>
    environment_id = read_environment_from_ns(rospy.get_namespace())
  File "/home/viki/catkin_ws/src/openag_brain/src/openag_brain/utils.py", line 39, in read_environment_from_ns
    "No environment id found in namespace \"{}\".".format(namespace)
ValueError: No environment id found in namespace "/".
Traceback (most recent call last):
  File "/home/viki/catkin_ws/src/openag_brain/src/openag_brain/software_modules/sensor_persistence.py", line 110, in <module>
    environment_id = read_environment_from_ns(rospy.get_namespace())
  File "/home/viki/catkin_ws/src/openag_brain/src/openag_brain/utils.py", line 39, in read_environment_from_ns
    "No environment id found in namespace \"{}\".".format(namespace)
ValueError: No environment id found in namespace "/".
[sensor_persistence_1-9] process has died [pid 12498, exit code 1, cmd /home/viki/catkin_ws/src/openag_brain/src/openag_brain/software_modules/sensor_persistence.py __name:=sensor_persistence_1 __log:=/home/viki/.ros/log/b6882eea-dcfb-11e6-918f-080027b46cdd/sensor_persistence_1-9.log].
log file: /home/viki/.ros/log/b6882eea-dcfb-11e6-918f-080027b46cdd/sensor_persistence_1-9*.log
[image_persistence_1-8] process has died [pid 12497, exit code 1, cmd /home/viki/catkin_ws/src/openag_brain/src/openag_brain/software_modules/image_persistence.py __name:=image_persistence_1 __log:=/home/viki/.ros/log/b6882eea-dcfb-11e6-918f-080027b46cdd/image_persistence_1-8.log].
log file: /home/viki/.ros/log/b6882eea-dcfb-11e6-918f-080027b46cdd/image_persistence_1-8*.log
[WARN] [WallTime: 1484688439.476432] Cloning into '/tmp/tmpceP0Kf/lib/openag_firmware_module'...
[WARN] [WallTime: 1484688442.985929] Cloning into '/tmp/tmpceP0Kf/lib/openag_dht11'...
[WARN] [WallTime: 1484688446.409180] Cloning into '/tmp/tmpceP0Kf/lib/openag_binary_actuator'...
```

LeonChambers, 2017-01-18:
The documentation for openag_python [here](http://openag-python.readthedocs.io/en/latest/) explains what it is and how it relates to this repo. There is no way you were running without it before, so you must have installed it at some point.

This new error is a result of #89. The fix is to set an environment in the configuration of the image_persistence and sensor_persistence modules (see #97). Reloading the default fixture will do this for you but will also reset any changes you have made to the other module configurations (if you have made any).

closed 2017-01-19 by LeonChambers