OpenAgricultureFoundation/openag_brain #108: Cache firmware module code

LeonChambers, 2017-01-25:
Right now, the flash script allocates a temporary directory to use for building the arduino code. This means that on every run, the script has to download the code for all of the enabled firmware module types and compile all of it. We could noticeably decrease this work and facilitate offline use cases by always building the Ardunio code in the same directory and only pulling firmware code when it has been updated.

davoclavo, 2017-01-30:
I looked into this and found that the flash script [already accepts](https://github.com/OpenAgInitiative/openag_brain/blob/master/scripts/flash#L49) a `--build_dir` argument flag to set the build directory (defaults to a `/tmp/` folder).

Please correct me if I'm wrong, but it seems that the flash module is only a utility made for testing/debugging. It was probably meant to be used by the `arduino_handler.py` software module, but [this is not currently the case](https://github.com/OpenAgInitiative/openag_brain/blob/master/src/openag_brain/software_modules/handle_arduino.py#L65-L73), as both independently run a Popen to `openag firmware run -p ros -t upload`

Anyways, as you mentioned, just setting the build directory to a fixed location will cache all the firmware module dependencies. I made [a small test hardcoding a folder in `arduino_handler.py`](https://github.com/davoclavo/openag_brain/commit/34e4e88a43aea57728a9035646826a4db5c77fb0)  and when the brain is restarted it doesn't re-download the dependencies.

I was considering adding it as a parameter when registering the `arduino_handler` instance, but I don't think this build_dir may change between runs, so the folder could just be defined at a static location (e.g. `~/openag_arduino`).

Spaghet, 2017-07-20:
We now version control the firmware modules in the openag_brain/firmware/lib directory so this no longer applies. Closing.

closed 2017-07-20 by Spaghet