OpenAgricultureFoundation/openag_brain #110: Change how we decide what is a sensor and what is an actuator

LeonChambers, 2017-01-25:
We currently have the simplifying assumption that all inputs to firmware modules are actuators and all outputs from firmware modules are sensors. This is limiting in a couple ways. First, some of the inputs to firmware modules are actually for calibration but are being treated as actuators, which is confusing. Second, it might be useful in the future to have software modules that act as sensors or actuators. The Raspberry Pi has plenty of GPIOs, so we might imagine migrating some (or even all) of the sensors from the Arduino to the Raspberry Pi. It would be natural for these devices to still be able to show up under `/sensors` and `/actuators`

#105 introduces a concept of categories that could be useful for solving this problem. Instead of assuming that all of the outputs from firmware modules are sensors, the system could assume that all of the outputs from modules (either firmware or software) with the category "sensors" are sensors. Similarly, all inputs with the category "actuators" would be treated as actuators. Inputs with the category "calibration" would be separated into their own namespace `/calibration`.

wsnook, 2017-01-26:
> The Raspberry Pi has plenty of GPIOs, so we might imagine migrating some (or even all) of the sensors from the Arduino to the Raspberry Pi.

I agree. As I've been studying the PFC2 code and hardware designs for the past couple weeks, I've been wondering about that too.

Here are some related questions to consider:

1. Are all the sensors compatible with the Pi's 3.3V GPIO? The RaspberryPi.org [GPIO docs](https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md) say, "*Connection of a GPIO to a voltage higher than 3.3V will likely destroy the GPIO block within the SoC.*"

2. Does the Pi 3 have enough hardware PWM? The PFC2 fixture is [set up to use three PWM actuators for the LEDs](https://github.com/OpenAgInitiative/openag_brain/blob/b75126b516187ae1f0c7212454c6ff0ed284f2ae/fixtures/personal_food_computer_v2.json#L7-L29). It looks like the Pi 3 has [two channels of hardware PWM](http://pinout.xyz/pinout/pin12_gpio18), but [they may not work reliably](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=150316).

3. If the Pi 3's hardware PWM isn't adequate, would it be reasonable to use [Wiring Pi's software PWM library](http://wiringpi.com/reference/software-pwm-library/) for the LEDs?

Coming at it from a different angle... based on the fixture, the PFC2 needs:
- 3 PWM outputs (LEDs)
- 8 binary outputs (pumps, heater, cooler, humidifier)
- 2 pins for an i2c bus handling four sensors (pH, EC, CO2, AM2315)

Thirteen pins isn't much. The [PFC2 hardware docs](https://github.com/OpenAgInitiative/openag_pfc2) also mention water level sensors and fans beyond what's in the fixture. But, that should all still fit on a Pi 3 and Arduino Uno with room to spare--using a Mega seems like overkill. Maybe the Pi could do relays and the Uno could do i2c, PWM, and miscellaneous sensors.

Instead of the Arduino being a ROS node, you could have a Python process talk to ROS. By implementing the sensor message protocols and calibration logic at the python layer, you could keep the Arduino sketch dirt simple--just a little REPL on the serial port providing a few basic I/O commands.

LeonChambers, 2017-01-26:
So the hardware configuration seems pretty unlikely to change in the near term. I think we're pretty set on RPi + Arduino Mega for the v2, but you'd have to ask someone on the hardware team to be sure (e.g. @jakerye). That being said, feel free to test out alternate configurations and let us know how it goes.

wsnook, 2017-01-26:
> So the hardware configuration seems pretty unlikely to change in the near term.

That makes sense. With the hardware as-is, it should still be possible to refactor firmware so that calibration is easier and the Mega doesn't need to be flashed frequently.

I've got a Pi 3 and Arduino Mega on order so I can run the code instead of just reading it. I'll put this issue on my list of stuff to look at. If I come up with anything potentially useful, I'll make a pull request with proof of concept code.

wsnook, 2017-01-27:
@LeonChambers I think I just found the reason why the Atlas pH and EC sensors need to be on a sensor board rather than just wiring them together on the same i2c bus.

Page 18 of Atlas Scientific's [pH EZO data sheet](http://www.atlas-scientific.com/_files/_datasheets/_circuit/pH_EZO_datasheet.pdf) says, "When reading pH and Conductivity together, it is strongly recommended that the EZO pH circuit is electrically isolated from the EZO Conductivity circuit... To correct this problem the power and data lines need to be electrically isolated."

That makes it sound like the Atlas pH sensor needs to stay connected to the Arduino Mega's existing sensor board shield.

closed 2017-06-26 by Spaghet