OpenAgricultureFoundation/openag_brain #129: Independent light control for red, blue, white

gordonbrander, 2017-02-14:
We want to have independent control within the light recipe for red, blue and white.

2 ways to do this that I'm aware of:

1. (easy) have separate environmental variables for `light_illuminance_red`, `light_illuminance_blue`, etc.
2. Describe illuminance in terms of `HSB` or `RGB` or `RGBK` (where k ~white LED) and translate that into some combination of red, blue and white panel.

I think #1 is easier, but also makes more sense, since panels aren't like displays... they can't create any combination of RGB. I could also see wisdom in a custom encoding scheme like: 

```javascript
{
  "type": "light_illuminance",
  "value": [RED, GREEN, BLUE, WHITE, UV]
}
```

Where each variable is a value between `0..255`. In this world, we would have a `light_controller` module that handles demuxing this. @LeonChambers any opinions?

LeonChambers, 2017-02-15:
I think we should go with the first option for now.

This requires us to set a definition of what specific wavelength to use for red and blue, which would probably just be whatever wavelengths our specific light fixtures produce. This does mean that if someone else is using a different light fixture, the environment that they are creating will be slightly different in a way not captured by the recipe, which is not ideal.

To be rigorous about this, we would have to actually represent the exact spectrum of light (intensity at every wavelength) that should be produced. Specifying a color is not enough because different light spectra can produce the exact same color but have different effects on the plants. Also, you can't encode infrared and UV lights in this way. Doing something this general would be pretty annoying in software though, and I don't think we should worry about it right now.

gordonbrander, 2017-02-15:
@LeonChambers I like your thinking here. We can always add a separate env var that encodes a more complete model at a later date.

gordonbrander, 2017-02-16:
Within https://github.com/OpenAgInitiative/openag_python/pull/35, added `light_intensity_red`, etc.

gordonbrander, 2017-02-16:
Closed by:

- https://github.com/OpenAgInitiative/openag_brain/pull/131
- https://github.com/OpenAgInitiative/openag_python/pull/35

closed 2017-02-16 by gordonbrander