OpenAgricultureFoundation/openag_brain #134: Ebb/Flow On/Off cycle controller?

Spaghet, 2017-02-20:
In aquaponics there is a need for something like a "5 minutes on, 10 minutes off" control for the circulation pump in order to conduct ebb/flow mechanics. Similar control should be useful in aeroponics as well.

Currently, one way to do this is the simple recipe doing 
[0, pump_var, 1], [300, pump_var, 0] repetitively, I think.
Is there an existing mechanic for this kind of control, or should I implement a software module to do this maybe?

LeonChambers, 2017-02-20:
You should just implement a software module to do this

Spaghet, 2017-03-05:
There seems to be a firmware module that does this called pulse_actuator but I haven't tried using it yet

closed 2017-05-02 by Spaghet