OpenAgricultureFoundation/openag_brain #178: usb_cam node crashes when device name changes

Spaghet, 2017-05-10:
This is related to #166 in that it is naiive to assume the camera will always be on `/dev/video0` and it will sometimes change to `/dev/video1`. This causes the node to crash like so:

```bash
[ERROR] [1494445358.751953273]: VIDIOC_DQBUF error 19, No such device
[environments/environment_1/aerial_image-16] process has died [pid 27197, exit code 1, cmd /home/pi/catkin_ws/devel_isolated/usb_cam/lib/usb_cam/usb_cam_node __name:=aerial_image __log:=/home/pi/.ros/log/03c461d6-35b4-11e7-8b52-b827eb8ab0fe/environments-environment_1-aerial_image-16.log].
log file: /home/pi/.ros/log/03c461d6-35b4-11e7-8b52-b827eb8ab0fe/environments-environment_1-aerial_image-16*.log
```

Spaghet, 2017-05-11:
There's a different error which requires you unplug the camera and replug it.
It give something related to `select timout` which seems to be a fairly common OpenCV/Linux bug

Spaghet, 2017-05-23:
https://raspberrypi.stackexchange.com/questions/12246/why-does-usb-port-enumeration-change 

Reading through this

Spaghet, 2017-05-23:
http://rolfblijleven.blogspot.nl/2015/02/howto-persistent-device-names-on.html

Using the hard-wired device ID to determine the /dev/ name seems to be the way to go.

ALxander19, 2018-12-31:
Thanks man! It help me a lot!