OpenAgricultureFoundation/openag_brain #181: Recipe gets forgotten

Spaghet, 2017-05-12:
@rbaynes has been manually restarting the climate recipe every morning on our demo bot.

Not entirely sure what the cause is, maybe the param servers only persist for a few hours, but one fix would probably be to hold "current_recipe" in the database and refer to it periodically.

gordonbrander, 2017-05-12:
@Spaghet does this happen after a reboot? Does it happen after a docker container restarts (following a crash)? Do we know when it happens?

This PR fixed a regression that was preventing `RECIPE_START` and `RECIPE_END` variables from being stored in the DB, therefore preventing recipes from picking back up across reboots or across docker container reboots (after a crash). https://github.com/OpenAgInitiative/openag_brain/pull/176

cc @rwdavis513 who has also seen this.

Spaghet, 2017-05-12:
I'm personally not sure what the cause of the problem is and haven't looked too deep into it. @rbaynes, can you share what is happening? Do you have to restart brain, or is it on the UI?

If the RPi is rebooting and restarting the process since we `systemctl enable`d it, this might be the cause.

gordonbrander, 2017-05-12:
Please note that an uncaught exception (which would cause the Docker container to reboot) could have also caused this before #176 was landed.

Spaghet, 2017-05-12:
I don't think we're using docker on that bot, but it's good to have that in mind

rbaynes, 2017-05-12:
@Spaghet we are using docker on the top demo-bot.

What I see as a user is:
A) no lights on when I get to the lab in the morning.
B) I check the UI and it says recipe: None.   I refresh the UI and it still says None.
C) I restart the general_greens two times.  The first time I get an error, the second time works.

gordonbrander, 2017-05-17:
There are 2 things that might be going on here:

1. Recipe might get forgotten on backend. This could be related to the fix here https://github.com/OpenAgInitiative/openag_brain/pull/176. We need to test.
2. On the UI-side, the initial state of the recipe picker says "None" when it should say "Loading". If the connectivity between the Pi and the UI is bad, you could end up in a situation where the recipe says "none" even if the Pi is in fact running a recipe. We should fix this.

Spaghet, 2017-05-17:
[image]

rosparam is still set, so this might be a UI-side.

actually, the UI is also correct (It briefly shows "none" but switches to "general_greens" in about 500ms)
[image]

Maybe it's just entering the night phase...?

novemberalpha, 2017-05-17:
FWIW I'm not seeing this on my machines. Sometimes, I have to load a recipe 5x to get it to overwrite the current_recipe (via UI only) but aside from that they stay persistent over reboots and whatnot.

Spaghet, 2017-05-17:
I'm going to close this issue and put it under the openag_ui

closed 2017-05-17 by Spaghet