OpenAgricultureFoundation/openag_brain #199: Debug water_potential_hydrogen firmware issue. Causing lose sync to arduino and raspberrypi (2d)

gordonbrander, 2017-05-12:
(Exported from Trello https://trello.com/c/99HgSuYq/999-debug-water-potential-hydrogen-firmware-issue-causing-lose-sync-to-arduino-and-raspberrypi-2d)

Spaghet, 2017-05-13:
Is this just the sensor being in UART mode? Or something more software side

rwdavis513, 2017-06-09:
I ran into the same issue with the nutrient pumps today. I need to verify but I'm guessing it is because the bottles were empty and therefore werent able to pump correctly. I suspect the pumps throw some kind of error if they arent able to pump. Need to investigate and see what happens when no bottle is attached.

Spaghet, 2017-06-13:
There's no such error case, it's probably just because the pump firmware is written incorrectly.
There are blocking calls to `delay()` in the doser pump and pulse actuator firmware code which has been fixed in the bespoke_serial firmware.

closed 2017-06-26 by Spaghet