OpenAgricultureFoundation/openag_brain #200: Deprecate openag_python (2d + 1d)

gordonbrander, 2017-05-12:
Notes http://wiki.openag.media.mit.edu/contributors/gordonb/deprecate_openag_python

(Exported from Trello https://trello.com/c/OaR2s1FK/1002-deprecate-openag-python-2d-1d)

---

On 2017-05-05T21:27:39.132Z, @gordonbrander wrote:

I&#x27;ve gone through openag_python and made a list of all the items we have to do to deprecate it. I&#x27;m calling this 2 days of work, + 1 day of testing.

---

On 2017-04-29T21:15:11.000Z, @bobdavis27 wrote:

That sounds good to me. Could you create separate cards for each of the tasks that need to be done to deprecate it entirely?

---

On 2017-04-28T21:19:40.860Z, @gordonbrander wrote:

@bobdavis27 it turns out this is more complicated than expected. openag_python has python dependencies that aren&#x27;t available to rosdep.

I think we should just try to deprecate openag_python instead.

gordonbrander, 2017-05-15:
I've issued a PR to add [Voluptuous data validation lib](https://pypi.python.org/pypi/voluptuous) to rosdistro https://github.com/ros/rosdistro/pull/14990. This would let us add it to our package.xml as a rosdep dependency.

rbaynes, 2017-05-31:
the rosdistro team suggests we try using the Jesse debian package `python-voluptuous` which is `version 0.8.2-1 `
Rob > try this.

rbaynes, 2017-06-05:
Gordon's changes are now in my branch: https://github.com/OpenAgInitiative/openag_brain/compare/deprecate_openag_python_rob

rbaynes, 2017-06-13:
https://github.com/OpenAgInitiative/openag_brain/pull/275

closed 2017-06-13 by rbaynes