OpenAgricultureFoundation/openag_brain #202: develop beaglebone brain proof of concept and roadmap

gordonbrander, 2017-05-12:
(Exported from Trello https://trello.com/c/wPGNFrJL/1010-develop-beaglebone-brain-proof-of-concept-and-roadmap)

wsnook, 2017-05-17:
@gordonbrander Do you have a citation or more details on "Beaglebone is also supposed to have a file system that prevents corruption in case of hard shutdown or bad write to memory"?

I did some googling just now, but I didn't find any reference to Beaglebone being somehow more resistant than Raspberry Pi to flash filesystem corruption. I did find mentions of people having SD card corruption problems on Beaglebone though. It also looks like there are a bunch of different distros that will work on Beaglebone. Do you know which one your source was talking about?

As far as I can tell, there are two common techniques for making embedded linux robust against power failures:
1. Use a power failure detection circuit with battery backup that lasts long enough for a graceful shutdown.
2. Use a readonly filesystem on the flash disk which mounts a ramdisk for RW stuff needed by the runtime. The obvious disadvantage here is that you can't save anything to the disk. Splitting between RO and RW partitions wouldn't necessarily keep you out of trouble, because there's no guarantee that the card's wear-leveling will respect your partitions. For all you know, it might decide to move some of your RO stuff to a new flash block as part of updating your RW partition, and it might corrupt the block due to a badly timed power failure.

gordonbrander, 2017-05-17:
> @gordonbrander Do you have a citation or more details on "Beaglebone is also supposed to have a file system that prevents corruption in case of hard shutdown or bad write to memory"? ...Do you know which one your source was talking about?

I wish I did! We heard this word-of-mouth from someone with a lot of technical expertise, but need to confirm it is true. The exploration here is mostly just to see if this is actually the case... No commitment or plan.

@rbaynes has been doing some stress-testing on the corruptibility of the Pi. The workarounds you mention are also something we've been looking into. Currently working on a battery solution.

wsnook, 2017-05-17:
> I wish I did! We heard this word-of-mouth from someone with a lot of technical expertise, but need to confirm it is true. 

Hmm... I suggest that before you do anything else with Beaglebone, it would be good to get specifics on which distro, which filesystem, and why it's better, including dates for when the information was accurate.

My Pi 3 with regular Raspbian Jessie Lite is using ext4 which is a good journaling filesystem--supposedly the journalling does well at recovering from power interruptions. I've seen people say other filesystems are marginally better for reducing wear on flash disks, but ext4 is definitely solid and battle tested (not like ext2 which corrupts easy). The Raspberry Pi Foundation people have done a good job with Rasbian, and it's based on Debian to begin with. But, things have come a long way since the original Raspberry Pi hardware. There were plenty of problems with Raspberry Pi in the past--it's possible what you heard was based on a comparison from years ago.