OpenAgricultureFoundation/openag_brain #206: Find better solution for EWMA filtering of boolean topics

gordonbrander, 2017-05-12:
Probably we should not be encoding everything under `/environments` to `Float64`. If we used `std_msgs` types, we could intelligently handle topic filtering per topic type.

(Exported from Trello https://trello.com/c/Y9JeYN0m/946-find-better-solution-for-ewma-filtering-of-boolean-topics)

---

On 2017-03-14T21:01:45.277Z, @gordonbrander wrote:

cc @bobdavis27 

---

On 2017-03-14T21:01:34.842Z, @gordonbrander wrote:

Original workaround solution: https://trello.com/c/OqXEJoTK/939-fix-water-loop-and-other-binary-binary-actuation-loops-by-fixing-ewma-topic-filter-for-boolean-topics

Spaghet, 2017-05-13:
Same as #70 probably.

gordonbrander, 2017-05-13:
Yeah it is.

closed 2017-05-13 by gordonbrander