OpenAgricultureFoundation/openag_brain #208: Fix dependency management and release process

gordonbrander, 2017-05-12:
The problem: we have release management and dependency management at several layers:
- Images (for the pi)
- Docker (for provisioning)
- git (for firmware system and docker)
- apt-get (for Docker, and for ros but only sometimes)
- pip
- bash scripts (for ros but only sometimes)

Obviously, this can cause issues with reproducible. Docker can solve that problem, but only if the sources you&#x27;re giving it are reproducible.

Some options:

- Monorepo + vendoring + git subtree
  - (would like to avoid vendoring where possible)
- Run an apt repository
- Other?

(Exported from Trello https://trello.com/c/2rEWA0aI/919-fix-dependency-management-and-release-process)

rwdavis513, 2017-05-15:
@gordonbrander this was taken care of with monorepo correct?

Spaghet, 2017-07-20:
Some of this has been solved but some of this hasn't been solved yet.

There is still some gunk in `scripts/install.sh` that needs to be sorted out by external parties (like rosdep) as well as some apt-gets like couchdb.