OpenAgricultureFoundation/openag_brain #210: Get cameras working (ros usb_cam)

gordonbrander, 2017-05-12:
Requires vendoring (or otherwise installing) `usb_cam` library.

Work:

- https://github.com/OpenAgInitiative/openag_brain_install_rpi/pull/17

A DEMO-ONLY branch is available here https://github.com/OpenAgInitiative/openag_ui/tree/wfp.

It relies on these scripts being installed in the pi&#x27;s crontab https://github.com/gordonbrander/pi_image_server

(Exported from Trello https://trello.com/c/xKLxM1fb/916-get-cameras-working-ros-usb-cam)

rwdavis513, 2017-05-15:
Camera is switching ports going from /dev/video0 to /dev/video1. Unclear what is causing the switch, there could be other issues with usb_cam as well.

wsnook, 2017-05-16:
Ideas:
1. Does the problem go away if you connect the cameras through a powered hub?
2. What's the amperage of the power supply you're running the Pi on, how much load is on it besides the Pi (any motors?), what are the current requirements for the cameras? How does the power supply's current rating compare to the sum of worst-case loads?
3. What do you see if you look in /var/log/syslog about usb and usb cameras? When I was messing around with a Logitech webcam--prior to getting totally fed up with it--it seemed to be connecting and disconnecting at the usb level a lot. I read some stuff about the kernel UVC drivers relying on reseting usb devices as a way of dealing with flakey camera firmware.