OpenAgricultureFoundation/openag_brain #217: openag_brain: prototype Python sensor ROS nodes (2d)

gordonbrander, 2017-05-12:
Based on work Jake did in https://github.com/OpenAgInitiative/openag_brain_box/, investigating how hard it would be to move all sensors and actuators to Python ROS nodes.

- Notes http://wiki.openag.media.mit.edu/contributors/gordonb/python_peripherals
- Working branch https://github.com/gordonbrander/openag_brain/tree/python_peripherals
- Jake&#x27;s source: https://github.com/OpenAgInitiative/openag_brain_box

(Exported from Trello https://trello.com/c/QROvSWqx/958-openag-brain-prototype-python-sensor-ros-nodes-2d)

---

On 2017-04-05T22:06:40.921Z, @gordonbrander wrote:

Issued PR for pylibftdi which is used to talk to the USB sensors in brain box. ROS core team merged it. https://github.com/ros/rosdistro/pull/14457

---

On 2017-04-05T15:03:05.097Z, @jakerye wrote:

use this: http://raspberrypi.stackexchange.com/questions/19765/rpi-gpio-as-a-high-speed-square-wave-generator but with 7us on &amp; off time (instead of 4) to generate 140khz signal to drive chiller. also need to change pin to 24 (instead of 4)

---

On 2017-03-27T22:56:28.646Z, @gordonbrander wrote:

The brain box depends on https://github.com/quick2wire/quick2wire-python-api. This has no entry in the apt repository for Debian/Raspbian/ROS. 

We&#x27;ll either have to  **vendor** it or **create an apt package**.

---

On 2017-03-27T17:05:05.589Z, @gordonbrander wrote:

https://github.com/OpenAgInitiative/openag_brain_box

Spaghet, 2017-07-20:
This looks like a duplicate of #249 but that one is for the brain box and this one is for the PFC so we should keep this open until we get a PFC with the Pi hat working

Spaghet, 2017-07-20:
hmm...That's not what I meant github :/