OpenAgricultureFoundation/openag_brain #224: Debug: ROS USB Cam and setup image processing (2d)

gordonbrander, 2017-05-12:
(Exported from Trello https://trello.com/c/OeOqMsZs/998-debug-ros-usb-cam-and-setup-image-processing-2d)

---

On 2017-04-27T16:25:17.958Z, @bobdavis27 wrote:

process[environments/environment_1/nutrient_flora_duo_a_controller_1-12]: started with pid [25345]
process[environments/environment_1/nutrient_flora_duo_b_controller_1-13]: started with pid [25381]
process[environments/environment_1/air_flush_on_controller_1-14]: started with pid [25392]
process[environments/environment_1/sensor_persistence_1-15]: started with pid [25455]
process[environments/environment_1/aerial_image-16]: started with pid [25460]
process[expand_diagnostics-17]: started with pid [25478]
process[arduino_handler-18]: started with pid [25529]
[ERROR] [1493310246.236244276]: VIDIOC_S_FMT error 16, Device or resource busy
process[api-19]: started with pid [25557]
[environments/environment_1/aerial_image-16] process has died [pid 25460, exit code 1, cmd /opt/ros/indigo/lib/usb_cam/usb_cam_node __name:=aerial_image __log:=/home/pi/.ros/log/eac0e91e-2b65-11e7-88c3-b827ebff173d/environments-environment_1-aerial_image-16.log].
log file: /home/pi/.ros/log/eac0e91e-2b65-11e7-88c3-b827ebff173d/environments-environment_1-aerial_image-16*.log

Spaghet, 2017-05-12:
Do you think fswebcam was trying to use the camera here?

gordonbrander, 2017-05-12:
@Spaghet it would have been ros usb_cam... does that use fswebcam?

Spaghet, 2017-05-12:
Well, if fswebcam was trying to use the camera, and ROS usb_webcam tries to use the device, it would probably be "busy"

gordonbrander, 2017-05-13:
Smart!

closed 2017-05-13 by gordonbrander