OpenAgricultureFoundation/openag_brain #235: Create a virtual test environment for code testing on a desktop or server instead of the pi

rwdavis513, 2017-05-16:
## Problem 
It slows down development to require testing on a Raspberry Pi only. Having a virtual environment which could run on a personal computer could speed up turn time of development.

https://developer.mozilla.org/en-US/docs/Mozilla/Developer_guide/Virtual_ARM_Linux_environment

rbaynes, 2017-05-16:
I will take this on.  I will also allow us to run a CI server that can be triggered on any merges to our repos.

And the possibility of just copying / installing the pre-built ARM binaries / images / packages to the pi.

gordonbrander, 2017-05-17:
- Either cross-compile on Linux machine
- OR if we end up on Ubuntu, this should make cross-compile easier

gordonbrander, 2017-05-17:
Investigate [Bloom](https://wiki.openag.media.mit.edu/ros/bloom)