OpenAgricultureFoundation/openag_brain #240: arduino-raspberryPi lose sync: ph pump command causing loss of sync

rwdavis513, 2017-05-16:
PH actuator (pump):
https://trello.com/c/99HgSuYq/999-debug-water-potential-hydrogen-firmware-issue-causing-lose-sync-to-arduino-and-raspberrypi-2d
Publish desired ph
loses sync
reconnects every 15-30s
Hypothesis:
maybe some infinite loop here? Do we have ph solution?
Could be blocking firmware?

wsnook, 2017-05-16:
Have you ruled out the possibility of an electrical problem? Switching inductive loads is an easy way to get glitches on your power supply rails. If you're getting minor brownouts or transients on the power to the Raspberry Pi, it could conceivably reset the Arduino without resetting the Pi.

To check for this with an oscilloscope, hook up a probe to GND and VIN on the Arduino and watch for voltage sags or transients. From what I read, you can also watch the Pi's power light ([if it goes out, that means the input voltage is too low](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=102814)) or hook up a monitor and [look for the yellow lightning bolt thing in the corner](https://www.element14.com/community/thread/57559/l/pi3-shows-yellow-lightning-bolt-with-official-power-supply?displayFullThread=true)

Spaghet, 2017-05-16:
@wsnook The Arduino is powered using a DC plug separate from the USB cable so this particular problem might be a different case.

wsnook, 2017-05-16:
Interesting... is there other stuff on that same power supply? Motors maybe? The same principle applies for looking for glitches with a scope whether or not the Pi is providing the power. I'm not 100% sure about which pins you'd need to watch, but if you put one channel on the input to the regulator and a second channel on the 5V output, it should be easy to tell if there are any power supply problems.

gordonbrander, 2017-05-17:
ph pump command has caused loss of sync before. This can also be caused by a number of factors:

- I2C command error
- Too many sensors/actuators
- Too many messages sent over topics
- No firmware/wrong firmware flashed on Arduino

Basically for nearly any problem with rosserial you get the same opaque error.

gordonbrander, 2017-05-17:
Duplicate of https://github.com/OpenAgInitiative/openag_brain/issues/167

closed 2017-05-17 by gordonbrander