OpenAgricultureFoundation/openag_brain #244: creating a Debian Blend and/or Live ISO image for OpenAg

dpocock, 2017-05-16:
Various other ecosystems (e.g. ham radio/SDR) have a Debian blend and live ISO image.  These are not the same thing but they are closely related.  A blend or ISO image provides a quick and easy way for new users to get up and running and it also provides a managed approach to upgrades using the Debian infrastructure.

This could be achieved by packaging parts of the OpenAg code base for the official Debian repositories or using a standalone repository until it becomes more stable.  The Debian experimental and backports repositories may also be relevant at different phases in the project.

Spaghet, 2017-05-16:
https://www.debian.org/blends/

Debian Blends are basically packaged forms of the entire operating system? I'm pretty sure we don't need to go as far as that to do releases. One possible thing we were thinking was to zip the entire ROS `catkin_ws` which someone will unzip in their home directory, and people could simply `$ source ~/catkin_ws/devel_isolated/setup.bash`.

dpocock, 2017-05-17:
Think of it from the user perspective:
- the software is available as standard packages that can be installed to an existing system
- one of your release artifacts is a ready-to-run ISO that people can burn onto a CD, USB stick or SD card

I agree that unzipping in the home directory is not so difficult, but there is a saying that for every extra step in the installation, you lose half the users (much like the saying that for every extra field on a form, some people won't finish completing the form).  Having the ISO image as a goal is very worthwhile.

I've previously used [live-build](https://packages.qa.debian.org/l/live-build.html)  but now there is also [live-wrapper](https://debian-live.alioth.debian.org/live-wrapper-doc/).  Using live-wrapper may be the way to go.

Spaghet, 2017-07-20:
One of the things we've been thinking of is just making a flashable OS image (.img file) as part of the release process. 
This seems to be fairly common practice in the Raspberry Pi world and I think that would be an acceptable way to go about releases. 

Debian Blends aren't on our to-do list though, so I'm going to close this one. If there's a compelling reason to use Debian Blends that we're missing I am welcome to suggestions.

closed 2017-07-20 by Spaghet