OpenAgricultureFoundation/openag_brain #246: (Exploration) Remove codegen, replace with versioned sketches for different HW configurations

gordonbrander, 2017-05-17:
## Summary

Replace separate firmware modules and codegen with versioned sketch files (one for each HW config).

## Motivation

- rosserial and issues with Arduino are the cause of most problems in the system at this point.
- reducing the complexity of this portion of the stack makes it easier to troubleshoot (less of a black box)
- Arduino sketches are familiar to makers and versioning the source of the Arduino sketch allows users to easily jump in and contribute.

## Details

- Keep firmware classes as-is for now
- Check in the sketch source code

## Drawbacks

- Source code duplication for different HW configurations.
  - Fix by keeping firmware classes and header files, allowing for code re-use

gordonbrander, 2017-05-17:
Rewrite arduino firmware, single control loop, no blocking.

wsnook, 2017-05-17:
This will probably make your life so much easier. If there's anything you want to swipe from https://github.com/wsnook/farm/blob/master/firmware/photon.ino, please do.

I made a little REPL thing for interactively polling sensor values, but it's also easy to automate. The big advantage is that you can manually test the exact same firmware that will be used by code further up your stack.

Spaghet, 2017-06-15:
Dealt with in #271, closing.

closed 2017-06-15 by Spaghet