OpenAgricultureFoundation/openag_brain #247: [Exploration] Replace rosserial with bespoke serial protocol

gordonbrander, 2017-05-17:
## Summary

Replace rosserial with a simple custom serial protocol (probably CSV) and a corresponding Python ROS node that knows how to interpret the protocol and publish the result to ROS topics.

## Motivation

rosserial has a number of issues:

- Opaque error messages. Multiple error cases result in same message. https://github.com/OpenAgInitiative/openag_brain/issues/167
- A limited number of topics is supported (and we are already up against the limit) https://github.com/OpenAgInitiative/openag_brain/issues/238
- The topic buffer is constrained and it is easy to spill over. The actual bandwidth for serial is waaaay more than rosserial allows.

As well as other issues https://github.com/OpenAgInitiative/openag_brain/issues/237https://github.com/OpenAgInitiative/openag_brain/issues/237

Most high-impact issues in the system are currently caused by rosserial. See https://github.com/OpenAgInitiative/openag_brain/milestone/3 for tracking these issues.

## Details

- Define a simple protocol that allows us to send info over serial to a ROS node
  - CSV or JSON
- Message (stringified)
  - int (control char representing pin or address or var type
  - float (value)
- Use newline char as EOM terminator (makes it work with serial monitor)

Questions:

- Send readings separately or together?
- Retain last state and send that?

Things to consider:

- Handle error cases
- Don't overflow Arduino serial buffer
  - Rate limiting?
- Baud rate
- Loop frequency
- Buffer strategy (don't overwhelm it)

## Drawbacks

...

## Alternatives

- Remove the Arduino altogether and use the GPIO pins on the SBC
  - This would require a wire harness or new signal board that fits the Pi's form factor

## Related

#202

gordonbrander, 2017-05-17:
Related #202, #246

wsnook, 2017-05-17:
In case it might give you some ideas, here's an example trace of the serial port protocol I'm using:
```
> help
 Commands:
  ds18b20  Get tank C with DS18B20
     tank  Get all tank sensor values
   si7021  Get air C & %RH with Si7021
   bme280  Get air C, %RH & hPa with BME280
      air  Get all air sensor values
   red on  Red LED on
  red off  Red LED off (default)
   yel on  Yellow LED on
  yel off  Yellow LED off (default)
     wifi  Get wifi status
  wifi on  Enable wifi for flashing firmware
 wifi off  Airplane mode (default)
     help  Show this help message

 To disconnect from screen, use ctrl-a k.
 For cu, use ~~. inside an ssh session, or
 use ~. locally.
> ds18b20
tank:C=26.1:sensor=DS18B20
> air
air:C=26.2:%RH=53.0:sensor=Si7021
air:C=26.6:%RH=51.1:hPa=964.7:sensor=BME280
> tank
tank:C=26.1:sensor=DS18B20
> si7021
air:C=26.2:%RH=52.9:sensor=Si7021
> bme280
air:C=26.6:%RH=51.3:hPa=964.7:sensor=BME280
> red on
> red off
> wifi
wifi:airplaneMode=true:RSSI=0
```

gordonbrander, 2017-05-18:
@Spaghet did a quick prototype of this approach for sensors using a simple CSV string message format that sends an status code along with an ordered list of messages from each sensor.

- The Arduino sketch sends CSV along with status code for debugging
- The ROS node is tightly coupled with the Arduino sketch.
  - It manages the Arduino and most of the smarts should exist at this layer.
  - It knows the CSV column order and how to parse the bespoke message format.
  - It's job is to parse out the message and publish status as well as values to ROS topics

CSV format is nice because it can also be hooked up to the Arduino serial monitor for testing.

wsnook, 2017-05-18:
@gordonbrander Looks cool.

CarrierDirectJoe, 2017-05-18:
I've been adding source ino files to my firmware libs exactly because I can't see into the std:msg payloads that come from the raspberry pi. It would be great to have a proper output to verify against before incorporating into a lib / ROS.

gordonbrander, 2017-05-18:
@CarrierDirectJoe proper output meaning strings printed to serial monitor?

novemberalpha, 2017-05-18:
Ya, sorry. It was hard to debug gc0011 because I couldn't read the ROS outputs to serial and didn't really have a debug harness or whatnot. So serial output was what I hacked around.

novemberalpha, 2017-05-18:
Also sorry for changing accounts. I try to respond on my personal account and not work. :-)

gordonbrander, 2017-05-19:
Prototype ROS node. It still needs to be able to handle short reads. https://github.com/Spaghet/test_serial/blob/master/nodes/arduino_handler.py

Spaghet, 2017-06-15:
#271

closed 2017-06-15 by Spaghet