OpenAgricultureFoundation/openag_brain #254: [Exploration] Stateless Recipe Interpreter

gordonbrander, 2017-05-22:
- Concept: create a stateless recipe interpreter
- Working branch: https://github.com/OpenAgInitiative/openag_brain/tree/recipe_stateless

## Rational

- The current recipe interpreter is rather complex and does not allow you to "rewind"... because it is stateful it can only seek forward.
- Because the recipe interpreter uses a blocking timer and is stateful, it is very hard to test

# Details

- Recipe interpreter should be a stateless function
  - Unit test interpreter
- Put the code that stores environmental data points in DB in a separate ROS node
  - This is in keeping with single responsibility principle
  - It also means we can turn on/off the database storage by including/removing from launch file

cc @rbaynes @Spaghet

Spaghet, 2017-05-22:
I like it, allows for recipe scrubbing as mentioned in #79  much easier.

wsnook, 2017-05-23:
@gordonbrander I don't understand what you mean by a stateless interpreter. As I understand interpreters, maintaining state is an essential part of what they do (e.g. call stacks, variables, current instruction being executed, etc.).

Do you mean you want to write a compiler that translates recipe source code into an intermediate representation--maybe a list of scheduled actions for changing control parameters? If so, that's a great idea. You could unit test the scheduler, unit test the modules that implement the configurable control loops, and unit test that recipe source files produce the correct compiled output.

gordonbrander, 2017-05-24:
@wsnook stateless in the sense that the recipe is read by a stateless function (https://github.com/OpenAgInitiative/openag_brain/blob/recipe_stateless/nodes/recipe_handler.py#L46), which is easy to test.

Previously, the recipe interpreter was a an iterator that made blocking calls to sleep.

wsnook, 2017-05-24:
Hmm... maybe I'm just getting tripped up by making assumptions that you're using the semantics I'm familiar with. I think of interpreter in the interpreter vs. compiler sense--like the python interpreter. It sounds like probably you mean something else by "recipe interpreter".

TL;DR never mind. I'll wait and watch.

gordonbrander, 2017-05-24:
interpreter ~= "thing that reads the recipe file and turns it into setpoints over time"

rbaynes, 2017-05-30:
Exploration done.  Code implemented on branch stateless_recipe for this issue: https://github.com/OpenAgInitiative/openag_brain/pull/255

Will be merged into master once we have a few more tests and the whole team reviews it.

closed 2017-05-30 by rbaynes