OpenAgricultureFoundation/openag_brain #264: Add Automatic Database compaction to Couchdb

rwdavis513, 2017-06-01:
If the PFCs are left running a long time, the disk space can fill up and cause the device to hault. Need to add automatic database compaction to decrease the disk space usage and add an alarm to notify disk space is reaching its maximum (separate issue needs to be created for the latter).

rwdavis513, 2017-06-06:
For the PFCs running through a docker image, we might also need to remove old docker images. This could have been the root problem of this issue.
[Removing old docker images](https://www.digitalocean.com/community/tutorials/how-to-remove-docker-images-containers-and-volumes)

CarrierDirectJoe, 2017-06-06:
Yep! Docker images, containers, and volumes (not the same) will fill your SD card super fast. I was unable to install global on an 8GB card after loading docker. Had to wipe. I use 32GB SD cards now.

rbaynes, 2017-07-31:
if DB compaction doesn't help, write a script to replicate couch DB to a server, then trim the env. data pt. DB.

closed 2017-08-03 by rbaynes