OpenAgricultureFoundation/openag_brain #305: Arduino stops communicating with Rpi after hours of operation.

goruck, 2017-07-10:
The Arduino firmware stops communicating with the Raspberry pi after several hours of operation. I'm running this on a PFC2.

I turned on tracing and get the following in "environments-environment_1-arduino_handler-18.log" after several hours of operation (running a recipe). 

```text
[rosout][DEBUG] 2017-07-08 16:51:31,554: 
TRACE> arduino_handler serial write 86 bytes: >0,0.0,0.0,False,False,False,True,True,False,0.0,True,True,True,0.0,0.0,0.0,False,True<
[rosout][DEBUG] 2017-07-08 16:51:31,557: 
TRACE> arduino_handler serial read: >0,89.40,25.40,402,25.87,0,0,3.83,0.00<
[rosout][DEBUG] 2017-07-08 16:51:32,554: 
TRACE> arduino_handler serial write 86 bytes: >0,0.0,0.0,False,False,False,True,True,False,0.0,True,True,True,0.0,0.0,0.0,False,True<
[rosout][DEBUG] 2017-07-08 16:51:33,557: 
TRACE> arduino_handler serial read: ><
```

You can see that that a blank message is read from the firmware. Once this happens, no more valid messages are received by the arduino_handler and the actuators stop operating and the sensors stop reporting new data. I checked the other ros logs and don't see anything strange apart from what I reported above. 

The USB ports looks ok and are stable. For example:

```text
pi@raspberrypi:~/catkin_ws/src/openag_brain $ lsusb
Bus 001 Device 006: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
Bus 001 Device 005: ID 05a3:9520 ARC International 
Bus 001 Device 004: ID 05a3:9520 ARC International 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~/catkin_ws/src/openag_brain $ 
```
And I don't see anything odd in dmesg. 

I checked the free memory on the Arduino using [this](https://github.com/McNeight/MemoryFree) and get about 5 kBytes free RAM, so the Arduino is not running low on memory which something I suspected.

I've been debugging this for a while now and can't seem to find the problem. I'll keep trying but please let me know what else I can try or if you need more information.

BTW, I do see occasional errors on the writes from the handler to the firmware but look to be unrelated to the read problem. For example in "environments-environment_1-arduino_handler-18.log":

```text
[rosout][DEBUG] 2017-07-08 15:59:55,554: 
TRACE> arduino_handler serial write 86 bytes: >0,0.0,0.0,False,False,False,True,True,False,0.0,True,True,True,0.0,0.0,0.0,False,True<
[rosout][DEBUG] 2017-07-08 15:59:55,560: 
TRACE> arduino_handler serial read: >2,Arduino,1,Invalid message 97 bytes: |0_0.0_0.0_False_False_False_True_True_False_0.0_True_True_True_False_True comma counts: 15 != 18|<
[rosout][WARNING] 2017-07-08 15:59:55,562: arduino_handler ERROR>  Arduino: Invalid message 97 bytes: |0_0.0_0.0_False_False_False_True_True_False_0.0_True_True_True_False_True comma counts: 15 != 18|
[rosout][DEBUG] 2017-07-08 15:59:55,564: 
TRACE> arduino_handler publish_time
[rosout][DEBUG] 2017-07-08 15:59:56,555: 
TRACE> arduino_handler serial write 86 bytes: >0,0.0,0.0,False,False,False,True,True,False,0.0,True,True,True,0.0,0.0,0.0,False,True<
[rosout][DEBUG] 2017-07-08 15:59:56,560: 
TRACE> arduino_handler serial read: >0,88.60,24.90,394,25.56,0,0,3.83,0.00<
```

It looks like the firmware deals with these Ok.

Spaghet, 2017-07-10:
We see that as well. My understanding was we caught this and deal with it properly but @rbaynes is working on that right now.

rbaynes, 2017-07-10:
Hi @goruck,
What branch are you running?  (master or develop) ?

goruck, 2017-07-10:
Hey, I'm running on develop.

rbaynes, 2017-07-10:
@goruck  Please pull the develop branch and see if you still have arduino issues.  I just merged https://github.com/OpenAgInitiative/openag_brain/pull/308

I also wrote up some notes on how to manually test the serial protocol and the arduino: https://wiki.openag.media.mit.edu/contributors/rob.baynes/arduino_test

goruck, 2017-07-10:
Will do, might be later in the week due to other commitments. Thanks!

arduino_test

goruck, 2017-07-16:
I pulled the latest develop branch and ran another test but sadly got the same result - Arduino stopped communicating with Rpi after hours of operation.

I'm now running a new test with pyserial class serial.Serial timeout set to 5 s (instead of 2 s).

goruck, 2017-07-16:
I found the problem. The read buffer was overflowing. Looks like the serial write before the read in the arduino handler affected how fast the read could be executed and over time the read buffer grew to the point it overflowed after which it couldn't be read again. Here are some traces using inWaiting to monitor the bytes waiting to be read:

```text
[rosout][DEBUG] 2017-07-16 07:46:26,394: 
TRACE> arduino_handler inWaiting bytes: >1128<
[rosout][DEBUG] 2017-07-16 07:46:26,399: 
TRACE> arduino_handler serial read: >0,86.20,24.10,508,25.37,0,0,3.86,0.00<
[rosout][DEBUG] 2017-07-16 07:46:27,392: 
TRACE> arduino_handler serial write 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,1.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-16 07:46:27,395: 
TRACE> arduino_handler inWaiting bytes: >1356<
[rosout][DEBUG] 2017-07-16 07:46:27,401: 
TRACE> arduino_handler serial read: >0,86.10,24.10,509,25.37,0,0,3.87,0.00<
[rosout][DEBUG] 2017-07-16 07:46:28,390: 
TRACE> arduino_handler serial write 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,1.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-16 07:46:28,393: 
TRACE> arduino_handler inWaiting bytes: >1356<
```

A normal response from the arduino is around 38 bytes, slow service of the read results in more and more bytes waiting to be read. In the example above 1356 bytes are waiting (growing from 1128) but readline() only grabs about 38 bytes at a time. Sooner or later the buffer fills up and then reads stop working altogether. 

I changed the serial read code to the following:

```python
# Read
# https://stackoverflow.com/questions/1093598/pyserial-how-to-read-the-last-line-sent-from-a-serial-device
if (serial_connection.inWaiting()):
    while True:
        # buf = serial_connection.readline()
        temp_buf = temp_buf + serial_connection.read(serial_connection.inWaiting())
        if '\n' in temp_buf:
            lines = temp_buf.split('\n') # Guaranteed to have at least 2 entries
            buf = lines[-2]
            temp_buf = lines[-1] # Keep possible partial read for next time
            break
```

This will ensure all the bytes that are waiting will be read. So far it seems to be working, see trace below where the bytes waiting to be read are monitored before and after the new read code.

```text
[rosout][DEBUG] 2017-07-16 08:59:14,620: 
TRACE> arduino_handler serial write 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,1.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-16 08:59:14,622: 
TRACE> arduino_handler inWaiting bytes: >38<
[rosout][DEBUG] 2017-07-16 08:59:14,623: 
TRACE> arduino_handler inWaiting bytes: >0<
[rosout][DEBUG] 2017-07-16 08:59:14,625: 
TRACE> arduino_handler serial read: >0,86.40,24.10,492,25.25,0,0,3.86,0.00<
[rosout][DEBUG] 2017-07-16 08:59:15,620: 
TRACE> arduino_handler serial write 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,1.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-16 08:59:15,622: 
TRACE> arduino_handler inWaiting bytes: >38<
[rosout][DEBUG] 2017-07-16 08:59:15,624: 
TRACE> arduino_handler inWaiting bytes: >0<
[rosout][DEBUG] 2017-07-16 08:59:15,626: 
TRACE> arduino_handler serial read: >0,86.40,24.10,492,25.25,0,0,3.87,0.00<
```

I'm running a long test now, if everything works then I'll submit a PR with the code changes.

rbaynes, 2017-07-16:
Thanks Lindo, I'm looking forward to the PR.

Please leave in any trace() code you have added.   The inWaiting() check is
great.

Thanks,
-rob

Thanks,
Rob Baynes

goruck, 2017-07-17:
I submitted the PR (its part of #294). The final code is as follows:

```python
# Read
trace('arduino_handler temp_buf: >%s<', temp_buf)
trace('arduino_handler inWaiting bytes: >%d<', serial_connection.inWaiting())
temp_buf = temp_buf + serial_connection.read(serial_connection.inWaiting())
if '\n' in temp_buf:
    lines = temp_buf.split('\n') # Guaranteed to have at least 2 entries
    buf = lines[-2] # Last full line read (earlier lines will be discarded)
    temp_buf = lines[-1] # Keep for next time, its either a partial line read or empty string
```

I've been running a test for about a day and half, all looks good from my end!

rbaynes, 2017-07-18:
Works great.  Bug fixed.  I have had two bots running for two days.

goruck, 2017-07-21:
Hey Rob, I'm seeing this issue again. Looks like the serial read is timing out but still trying to figure out what's going on. Can you reopen the issue?

rbaynes, 2017-07-21:
Sorry Lindo, I closed this too early.   I have 3 PFCs running continuously and 2 are fine.  One of them seems to stuck not getting any data from the arduino.  Here is what I see after it ran fine for 3 days:
```
TRACE> arduino_handler temp_buf: ><
TRACE> arduino_handler inWaiting bytes: >0<
TRACE> arduino_handler serial read: ><
```

goruck, 2017-07-21:
Yeah seeing the same thing. I have some theories that I will investigate
today and hopefully one will result in a fix.

goruck, 2017-07-24:
Here's an update. Net is that I think I've fixed it for good this time but still don't have an absolute root cause and enough test run time and so the issue should be kept open for now. I'll submit a PR with the latest changes.

There are at least two problems here:

1. Since the Arduino writes error messages back to the Raspberry Pi before it writes sensor data and only one line is read from the buffer at a time, the serial read buffer will eventually overflow causing unpredictable results. Most of the errors I've seen are due to bad writes to the Arduino, they are not uncommon. This is the issue I identified and fixed earlier with the code that read all lines from the buffer but discarded all but the last one read. (I found a simpler way to handle this by using the pyserial flushInput() method after readline() which now replaces the code that used to handle this error.)

2. As we know, writing a message to the Arduino causes it to return data to the Raspberry Pi, but reading this data immediately after the write looks to be problematic. I'm not sure of the root cause, its likely some kind of race condition. See oscilloscope screenshots below. I added a delay after the write using rospy.sleep(arduino_delay_s), where arduino_delay_s = 0.5 / serial_rate_hz. The delay makes the read occur well after the Arduino writes the data to the Raspberry Pi.

With the two fixes listed above, the system looks to be very stable. Its been running for over a day but more testing is required.

I also made some changes to the write section in arduino_handler, specifically made the writes non-blocking and added flushOutput(). My thinking here was to ensure that the writes to the output buffer never timed out but I didn't see evidence of that. So I'm not sure these changes made a difference. 

In order to get some visibility into the Arduino firmware, I put some debug code in that toggles two digital pins that I'm monitoring with an oscilloscope. One pin goes high when a new message is received by the Arduino and goes low when its reset, the other goes high when a message is being transmitted from it (from the time serial prints start to after the buffer is flushed.) Note that 3 different types of messages can be transmitted, the sensor data, an error indicating a bad message was received and an error indicating some problem with the sensors. If the system crashes again, the scope should help me track down exactly what died. Below are some interesting screenshots from the scope. The upper trace is the new message from the Arduino, the lower is the start to end of a message transmission from the Arduino. 

Normal messages being received and transmitted (note 1 Hz rate):
[image]

Sensor data being transmitted:
[image]

Error data followed by sensor data being transmitted:
[image]

Strange pattern caused by a very long (~190 bytes) bad message being received:
[image]
Note the long time (at least 3 s) without a message being received by the Arduino, the fast messages received (2 Hz rate) and then the return to a normal write frequency (1 Hz rate). I'm not sure why this is happening, something may have timed out, perhaps the serial port or a read, but the read timeout is only 2 s. I suspect that without the rospy.sleep delay, this would lock up the system. This may be a root cause, will verify. The fast messages may be rospy.Rate's way of trying to keep a 1 Hz loop rate on average, if so really cool!

goruck, 2017-07-24:
Saw it again, the communication stopped after about 30 hours. From the Arduino scope traces it looks like it stopped receiving new messages from the Raspberry Pi. So, I'm assuming something must have stopped the serial writes from the arduino_handler. I changed the code to blocking writes with the same timeout as the reads (2 s at the 1 Hz loop frequency) and added an exception to trap write timeouts to see If this is actually happening. If I can catch it then maybe some recovery mechanism can be added. 

Also, I wanted to see if changing the baud rate to 9600 would make a difference. But making those mods to the Arduino firmware and the arduino_handler didn't work. Is there another place in the code where the serial baud rate needs to be changed?

I also changed arduino_delay_s = 0.25 / serial_rate_hz.to see how it affected the recovery time after the Arduino receives a very long bad message as noted above. Here's a screenshot:

[image]

It does look like the fast messages following the long delay (3.2 s) are rospy.Rate's way of trying to keep a 1 Hz loop rate on average. Still not sure what is causing the long long delay, something must be timing out and if so maybe this what eventually causes the system to stop communicating.

rbaynes, 2017-07-24:
Thank you for all the great work.  I'm out of the lab his week, without a
computer.   From memory, I think there are only two places to set the baud
rate.  One in src.ino and one in arduino_handler.py.

-rob

Spaghet, 2017-07-24:
Thanks for the detective work!
I think there's a fundamental flaw to how I wrote this node which stems from my lack of understanding of the UART protocol when I was thinking about how to go about implementing this.

Parsing only the last line was a bad idea. There should be a way to read all lines and act accordingly. In the initial implementation it assumed that you could read from the serial bus at any point and it would block until it reached a newline character using serial.readline() and that reading and writing timing wasn't so critical so we could implement them assuming they were like async streams but that doesn't seem to be the case. 
We changed it to do a request-reply type synchronization and kept the same parsing structure there which is displeasing anyways so if it's causing problems I'm open to changing it. 

I don't see the particular issue you see but I do see another issue in the Arduino firmware that's making it hard for me to debug this one..thank you again for your hard work!

Regarding baud rate:
There's a "baud_rate" param in the launchfile 
https://github.com/OpenAgInitiative/openag_brain/blob/develop/launch/personal_food_computer_v2.launch#L74

goruck, 2017-07-25:
Thanks for the info! I did try to change the baud rate in the launch file and the Arduino .ino file to 9600 but still couldn't get it to work. I don't think that's important anyhow, 115200 should be fine and is preferable.

Here's an update: 

I've been monitoring the serial TXD and RXD lines on the Arduino via the openag signal board. These are TTL level lines from the USB chip. I've seen what looks like an error condition in the Arduino where it doesn't send out data in response to a message from the rPi. This probably causes a read timeout in the arduino handler which in turn delays the next message. Here's a screenshot from the scope (top TXD, bottom RXD):

[image]

Its possible that the delayed write could cause a timeout. I changed the writes to have a timeout as indicated above which seems to allow the arduino handler to recover. The system has been running for about 30 hours now without issue. I'll keep monitoring.

In summary, here's what I have now which so far looks Ok for me:

1. Changed writes to have a timeout (same as reads)
2. Added a rospy.sleep between writes and reads.
3. Reads only take the first line in the buffer, other data is discarded.

goruck, 2017-07-25:
Forgot to add that when the error condition captured above happens I see the following messages:

```
TRACE> arduino_handler serial write 89 of 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,0.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-25 01:16:59,064: 
TRACE> arduino_handler serial read: >0,93.80,24.40,357,25.69,0,0,3.90,0.00<
[rosout][DEBUG] 2017-07-25 01:16:59,811: 
TRACE> arduino_handler serial write 89 of 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,0.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-25 01:17:02,065: 
TRACE> arduino_handler serial read: ><
[rosout][DEBUG] 2017-07-25 01:17:02,081: 
TRACE> arduino_handler serial write 89 of 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,0.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-25 01:17:02,340: 
TRACE> arduino_handler serial read: >2,Arduino,1,Invalid message 175 bytes: |0_0.0_0.0_False_False_False_False_False_False_0.0_True_True_Tru0_0.0_0.0_False_False_False_False_False_False_0.0_True_True_True_0.0_0.0_0.0_False_False comma counts: -1 != 18|<
[rosout][WARNING] 2017-07-25 01:17:02,342: arduino_handler ERROR>  Arduino: Invalid message 175 bytes: |0_0.0_0.0_False_False_False_False_False_False_0.0_True_True_Tru0_0.0_0.0_False_False_False_False_False_False_0.0_True_True_True_0.0_0.0_0.0_False_False comma counts: -1 != 18|
[rosout][DEBUG] 2017-07-25 01:17:02,351: 
TRACE> arduino_handler serial write 89 of 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,0.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-25 01:17:02,604: 
TRACE> arduino_handler serial read: >0,93.80,24.40,357,25.69,0,0,3.91,0.00<
[rosout][DEBUG] 2017-07-25 01:17:02,810: 
TRACE> arduino_handler serial write 89 of 89 bytes: >0,0.0,0.0,False,False,False,False,False,False,0.0,True,True,True,0.0,0.0,0.0,False,False<
[rosout][DEBUG] 2017-07-25 01:17:03,064: 
```

You can see the empty string read at 01:17:02,065 from the missed TXD and then the Arduino reports it received an invalid message which looks like two partial messages concatenated together.

Spaghet, 2017-07-25:
You are awesome 🔥 

I'll merge in your PR by end of day today if it's ready

goruck, 2017-07-25:
Hold off until tomorrow, I want to check a few more things. I think we're getting close on this one!

goruck, 2017-07-26:
Update:

I think I found another root cause. As noted above I've seen cases where the Arduino does not TX back to the rPi and after that the first TX it sends indicates it got a bad message from the rPi which look like two partial messages concatenated together, the first without a newline. Tracing through the Arduino sketch I can see how this could happen because the serialEvent() function is left with a partial message which gets added to the next time there's data in the HW RX serial buffer and serialEvent() runs again. Now, if there are 5 or 6 consecutive occurrences of partial messages sent by the rPi then the message buffer will overflow (its set to 500 bytes long and each message from the rPi is about 90 bytes). To prevent this condition, I changed the serialEvent() function as follows:

```cpp
void serialEvent() {
  if(stringComplete){
    resetMessage();
  }
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString but first check for potential overflow:
    // (this can happen if a few partial lines are recieved one after the other without newlines)
    if (sizeof(message) == (MESSAGE_LENGTH - 1)) {
      message += '\n'; // 1 byte add!
      stringComplete = true;
      return;
    }
    message += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') {
      stringComplete = true;
      return;
    }
  }
}
```

I'm going to give this a try with the other changes noted above. Also, I'll PR the latest changes but let's hold off merging them until I get some more run time. So far, even without the Arduino sketch changes, things are looking really stable!

Spaghet, 2017-07-26:
👌 🔥 💯

goruck, 2017-07-28:
# Update

My system is still looking good after several days of operation! I think this is ready to go. I've updated the PR (#317) with the latest changes and you can merge them if you think they are Ok.

While this is all still fresh in my mind, I recorded the history of this issue below in the form of a "5 Whys" root cause analysis. 

# Five Whys for "Arduino stops communicating with Rpi after hours of operation"

Q1: Why did the Arduino stop communicating with the Raspberry Pi after hours of operation?

A1: Nothing could be read from the Arduino in the arduino_handler in the Raspberry Pi. 

Q2: Why couldn't anything be read from the Arduino?

A2.1: The serial read buffer reached an overflow condition (or)

A2.2: The Arduino stopped serial transmission.

Q3.1: Why did the serial read buffer reach an overflow condition?

A3.1: (Root Cause # 1) The Aurdino writes first error messages (if present) and then sensor messages. But the read only takes the first message and will leave the second message in the buffer. Over time this causes many messages to be left in the buffer eventually leading to an overflow condition.

(Fix # 1) The code was modified to flush the read buffer after the first read, thus ensuring nothing is left in the buffer. 

Q3.2: Why did the Arduino stop serial transmission?

A3.2: If the Arduino receives an partial message without the newline delimiter the global message variable is never reset. Subsequent partial messages get appended and eventually the message buffer will overflow (its 500 bytes). When the buffer overflows memory corruption occurs and the Arduino sketch crashes stopping serial transmissions (and everything else).

(Mitigation # 2) An overflow check was added to sketch to ensure the message variable never exceeded 500 bytes.

Q4: Why does the Arduino receive partial messages without newline delimiters?

A4: This is not clear. The USB physical layer should be quite robust with very low BER, thereby making the probability of a message error very low (<< 1%). However it is observed that the message error rate is around 1%. Thus, its probably an upper layer problem, possibly Raspberry Pi serial write time outs caused by a low level USB driver issue or an issue in the USB HW itself.

(Mitigation # 3) Serial writes were changed from blocking to timed and exception handling added to log write timeouts (however none have been observed).

(Mitigation # 4) A ros rate delay was added between the Raspberry Pi write and read to slow down overall USB activity.

(Mitigation # 5) The write buffer is flushed after every write (but this only clears the OS buffers and probably not the HW USB buffers). 

These mitigations have reduced the occurrence of multiple back-to-back partial messages sent from the Raspberry Pi to the Arduino. However there is no guarantee that they won't occur but Mitigation # 2 provides a safety net.

In summary this issue has not been fully root caused but with the one root cause fix and the mitigations in place the system is very stable and there is very high confidence this issue will not resurface.

# Misc Notes

## Pyserial incorrect write time out behavior

 The pyserial doc states "writeTimeout = 0: non-blocking mode, return immediately in any case, returning zero or more, up to the requested number of bytes". However inspection of serialposix.py reveals the following:

```python
if self._writeTimeout is not None and self._writeTimeout > 0:
            timeout = time.time() + self._writeTimeout
        else:
            timeout = None
```

Thus, 0 and None result in the same behavior (writeTimeout=None) which is a blocking write with no time out.

## USB electrical integrity

Just to be sure I did swap out the USB cable between the Arduino and rPi to rule out a bad cable. It was fine. I also checked the electrical noise levels on the Arduino and rPi power supplies, they look good.

commit 3253b890 (2017-07-31): Merge pull request #317 from goruck/develop

add delay between writes and reads, other misc to fix #305
commit 78f326d1 (2017-07-31): add delay between writes and reads, other misc to fix #305
closed 2017-07-31 by rbaynes