OpenAgricultureFoundation/openag_brain #316: recipe_handler does not start up (appears to hang) if the environmental_data_point database is > 5G

rbaynes, 2017-07-21:
The simple solution if you don't care about your data is to delete the environmental_data_point DB and restart the brain.

The issue is that the by_variable query we are doing to find the last start and end recipe topic message time stamps is taking too long when the database is large.

The solution is to add two views to specifically return the timestamp of the last recipe_start and recipe_end message (and the recipe name) so we can decide if we should (and where) restart a recipe when the brain is starting up.

rbaynes, 2017-07-21:
https://blog.couchbase.com/understanding-grouplevel-view-queries-compound-keys/

Spaghet, 2017-07-24:
We can probably just add the `stale=ok` query parameter and it should be much faster.
Try this:
http://{ip}:5984/environmental_data_point/_design/openag/_view/by_variable?stale=ok&startkey=[%22environment_1%22%2C%22desired%22%2C%22recipe_start%22]&endkey=[%22environment_1%22%2C%22desired%22%2C%22recipe_start%22%2C{}]&grouping_level=3

The problem is that there are documents being uploaded (i.e. the view is being updated) as we query the view so it takes something like 22 seconds even for a close to empty database.

Adding the `stale=ok` query parameter shortens that request time quite a lot.
[image]

Spaghet, 2017-07-24:
The problem is that it no longer guarantees that a view is up to date, but we can ensure the view is being updated periodically and make sure it's frequent enough that we can just manually restart the recipe if we don't update the view before openag_brain crashes.

Spaghet, 2017-07-24:
So now the issue is "where do we put this code to periodically trigger the updating of the view".

Actually, since the recipe_start should really only be relevant for the message for recipe_start, we can just do `stale=update_after` and we will be ok.

rbaynes, 2017-07-24:
Thanks for all the work on this Rikuo!

PR #318 (2017-07-31): Add stale option in db query for recipe restarts
Fix #316 

Works on valkyrja/spaghet-a-bot nicely
closed 2017-07-31 by rbaynes