OpenAgricultureFoundation/openag_brain #320: AM2315 Unknown Function Code 255

Spaghet, 2017-07-24:
The AM2315 sometimes gets into a state where it just reads "function code 255" and doesn't recover.
I'll have to look into the sensor datasheet to see how it works but the Atlas sensors have a code 255 as well. It might also just be the sensor pulling HIGH for the entire duration of the message causing it to do this as well.

Spaghet, 2017-07-27:
I tried unplugging the GND pin on an AM2315 while the firmware is running and it looks like that reproduces this error. 
I'll look into this a bit more and see what we can do to either prevent this from happening or recover gracefully.

Spaghet, 2017-07-27:
I think I've been able to reproduce this error by unplugging the GND pin(probably any pin) on the AM2315 and then replugging it in. This causes the Arduino to hang and then only successful reboot when you fully unplug the AM2315 or Arduino.

rbaynes, 2017-08-03:
I also get this happening on rob-bot.   I need to put a new connector on the sensor and hopefully that will solve the issue.

Spaghet, 2017-08-03:
The thing I'm curious about is why it started happening when it didn't before. Either the other stuff was crashing before this would happen, or we changed the firmware to no longer be resilient to the change. Not sure.

goruck, 2017-08-03:
I see this occasionally as well but it doesn't cause the Arduino to lock
up. It goes away when I re seat (but do not unplug) the connection to the
signal board. I had assumed it was due to a marginal connection somewhere
in that cable but since you guys are seeing it too, maybe not. I'll
investigate a little more.

rbaynes, 2017-08-04:
This happened again in less than 12 hours on rob-bot.   I tried pushing the restart button on the arduino shield and it didn't help.  I unplugged the AM2315 and tested it on a different arduino connected to my laptop.  It worked fine.  Wiggled the connector and wires, no issues.   Plugged it back into rob-bot and all is fine.  

Perhaps there is a bug where that sensor shuts off and doesn't turn back on unless it is power cycled?

rbaynes, 2017-08-07:
Found rob-bot in the "arduino not sending any data" state that is the result of getting the 255 error for too long this morning.   AM2315 tests fine on the bench.   Tinned the leads and put a new connector on it, reinstalled and its fine again.  But for how long?

rbaynes, 2017-08-08:
I'm going to make a change to the arduino_handler that will detect if it is not reading any data. After a certain number of empty reads (perhaps 10, since 1 is normal with the buffer reset logic) I will reset the serial port, which seems to reboot the arduino and get everything working again.

I know this isn't the same problem as getting `TRACE> arduino_handler serial read: >2,AM2315 #1,1,Sensor responded with the wrong function code:255<`.

I will also detect and count those, since I was getting them on rob-bot before I made a new connector for the AM2315.

rbaynes, 2017-08-08:
Added these changes to an outstanding PR: https://github.com/OpenAgInitiative/openag_brain/pull/329