OpenAgricultureFoundation/openag_brain #328: ROS arduino_handler.py node crashed when serial port not available

rbaynes, 2017-08-04:
I thought we had solved this problem, but apparently it is back.  Or maybe we need to catch this exception too...

```
[WARN] [WallTime: 1501863601.753149] write failed: [Errno 5] Input/output error

[WARN] [WallTime: 1501863601.821312] could not open port /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85531303231351F09081-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_85531303231351F09081-if00'

Traceback (most recent call last):
  File "/home/pi/catkin_ws/src/openag_brain/nodes/arduino_handler.py", line 389, in <module>
    serial_connection.write(message) # Write len(message) bytes or timeout
AttributeError: 'NoneType' object has no attribute 'write'
[ERROR] [1501863603.919463867]: VIDIOC_DQBUF error 19, No such device
[environments/environment_1/arduino_handler-18] process has died [pid 2183, exit code 1, cmd /home/pi/catkin_ws/src/openag_brain/nodes/arduino_handler.py
```

rbaynes, 2017-08-07:
The fix for this, in PR #329 has been running for 3 days on the machine it crashed on.  Looks like the serial port switching issue is now handled.

Spaghet, 2017-08-12:
That's weird, considering we explicitly solved this a while ago..
Good to see it's patched