OpenAgricultureFoundation/openag_brain #342: no data received from sensors

LFri, 2017-10-26:
We've installed the software according to the [instructions](https://github.com/OpenAgInitiative/openag_brain_docker_rpi), updated the firmware, but we are not getting any data from the sensors.  We are not getting output from "rostopic echo" or any errors from "docker logs openagbraindockerrpi_brain_1" that we can see.  Are we missing anything in the logs or software?  Suggestions?? Thank you!!

[image]

[image]

Spaghet, 2017-10-27:
I'm pretty sure we haven't updated the docker image in a while. 
https://hub.docker.com/r/openag/rpi_brain/tags/ (7 months ago)

If you want to install openag_brain, I think we currently have a release on master https://github.com/OpenAgInitiative/openag_brain/releases/tag/v1.0.0

running `openag_brain/scripts/install_dev` should set things up (athough I've heard there are issues with the latest version of raspbian)

Alexis333, 2017-10-30:
@LFri  Had the same issue, just install the developer setup: https://wiki.openag.media.mit.edu/contributors/rob.baynes/rpisetup and use the mentioned Raspbian version otherwise it won't work correctly.

If you still don't have any data coming in, just write back.

coopmaster, 2017-11-03:
I am working with @LFri on this project and I followed the instructions (including using raspbian Jessie) and now we are getting a 500 error when trying to start a recipe with the openag_ui. When I try to do it manually from the command line with `rosservice call /environment_1/start_recipe long_test_recipe`, I get 

> ERROR: Unable to communicate with master!

It seems like ros isn't running, so I tried running `rosrun openag_brain main -f default` but it gives the error

> main: error: unrecognized arguments: -f

I also tried running the init_dev script, which gives

> Installing binary dependencies with rosdep
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_comm: No definition of [roslisp] for OS [debian]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
Finished setting up development environment. Ready to build with catkin_make.

goruck, 2017-11-26:
Hi,

I've successfully been using the openAg brain software on a Raspberry Pi. I've written a short installation howto which can be found [here](https://github.com/goruck/foodcomputer#installing-openag_brain-openag_brain-on-the-food-computers-raspberry-pi). Hope this helps, feel free to contact me directly for additional help. 

Thanks.

Lindo

coopmaster, 2017-11-30:
Ok, I reinstalled the os on the pi and reinstalled the software following @AxelStefan20's instructions https://wiki.openag.media.mit.edu/contributors/rob.baynes/rpisetup (which is pretty much the same as the one @goruck linked). After I got done, it still wasn't working. I am getting an error message when I try to manually try to run it using `rosrun openag_brain main personal_food_computer_v2.launch` (after I flash the arduino)

> [WARN] [WallTime: 1512077530.409973] arduino_handler ERROR>  AM2315 #1: Sensor responded with the wrong function code:255

I have no idea what this error message means though. Does it mean that I have the air temp/humidity sensor plugged into the wrong place? Is the documentation wrong for where to plug the sensors in? Right now I have it plugged into A0, and looking at the firmware it says `_i2c_address = 0x5c` but I don't know what that means.

goruck, 2017-12-01:
Hey,

I've seen this error as well and in my case it was from a bad connector from the sensor to the Arduino. Also see #320 for some other related info. I can check later if my sensor is plugged into A0 but I'm fairly certain it is.

Lindo

coopmaster, 2017-12-16:
I figured out our problem. I was trying to use the current software on hardware we got from the 1.0 bill of materials. So instead of having the AM2315, we have the AM2302. So we need to decide if we are going to upgrade our hardware or revert back to the 1.0 software. 

Thanks for everyone's help.

coopmaster, 2017-12-16:
We decided to upgrade the hardware to the latest version, 2.1. We were wondering if there was an upgrade list for the parts we would need to order for us to be able to utilize the latest software.